Klipper3d/klipper #2921: Invalid literal for float()

okkman, 2020-05-26:
First time Klipper user so I'm sure it's an ID10T problem but I can't seem to figure it out.  I have gotten as far as correct motion on all steppers, and individual homing but I am unable to successfully add additional features from the example_config items.  Could this be caused by cut and paste errors from that file to my printer.cfg?  If it is not a syntax issue it appears to be python related in that scripts attempting to retrieve data from the config fields appear unable to do so.

Any ideas?

Thank you.

[klippy.zip](https://github.com/KevinOConnor/klipper/files/4683328/klippy.zip)

jakep82, 2020-05-26:
It looks like you have a syntax issue with your [respond] section.  This is what your config looks like now:

```
[bed_mesh]
speed = 100
mesh_min = 40,10
mesh_max = 210,150
	[respond]
	default_type: echo
	default_prefix: echo:
```

This is what it should look like:

```
[bed_mesh]
speed = 100
mesh_min = 40,10
mesh_max = 210,150

[respond]
default_type: echo
default_prefix: echo:
```

okkman, 2020-05-26:
@jakep82 thank you!  I commented out those lines in my config and the parsing errors went away.  I'm not sure how the extra indentations got in there, but they do appear to be in the unmodified extras-config with my installation.  Thanks for your sharp eye!

closed 2020-05-26 by okkman