Klipper3d/klipper #2922: Printer not ready, can't fix it.

CharlesStArnault, 2020-05-26:
I heated my hotend to change the nozzle and to clean the hotend. When I removed the nozzle and started cleaning, i got the expected message saying extruder not heating at expected rate then It stopped heating. I finished cleaning and now I can't use my printer because It keeps saying printer not ready, i haven't made any modifications...

Please help. Thanks

[klipper.txt.txt](https://github.com/KevinOConnor/klipper/files/4684215/klipper.txt.txt)

jakep82, 2020-05-26:
If firmware_restart isn't working, you might need to add `restart_method: command` to your `[mcu]` section.

KevinOConnor, 2020-06-01:
The host isn't finding the micro-controller - you should double check the wiring and that it is flashed correctly.  If that doesn't fix it you should upgrade to the latest version of the software (see the FAQ for upgrade instructions) and try again.

-Kevin

KevinOConnor, 2020-06-14:
I'm going to close this as it looks like the original question was answered (at least as best as we could).

-Kevin

closed 2020-06-14 by KevinOConnor