Klipper3d/klipper #2924: Internal error on command:"G1"

Oranginator, 2020-05-27:
Hi guys,

i have no idea why I'm getting the error "Internal error on command:"G1"". Perhaps somebody is able to help me, thanks in advance!

Best regards Steffen
[klippy.log](https://github.com/KevinOConnor/klipper/files/4688547/klippy.log)

Hywelmartin, 2020-05-27:
#2793

Oranginator, 2020-05-27:
> #2793

I'm already using the latest build. I also updated my host software running on the SKR 1.1 Pro. With "make clean" etc.

KevinOConnor, 2020-06-01:
There was an error in step flushing in certain cases where SET_PRESSURE_ADVANCE was called after a G4.  I think this should be fixed now (commit 9850dc75).

-Kevin

Oranginator, 2020-06-03:
> There was an error in step flushing in certain cases where SET_PRESSURE_ADVANCE was called after a G4. I think this should be fixed now (commit [9850dc7](https://github.com/KevinOConnor/klipper/commit/9850dc75f4a7641a5bd12d9d2ccb7ca698619c88)).
> 
> -Kevin

Thank you, I'll try.

KevinOConnor, 2020-06-14:
I'm going to close this as it is believed the Klipper bug is now fixed.

-Kevin

closed 2020-06-14 by KevinOConnor