Klipper3d/klipper #2925: Multiple heatbeds [heater_bed] support?

Tomblarom, 2020-05-27:
So, I'm primarily using a 300x300mm bed for smaller prints. If I want to print anything bigger, I install my 400x400mm bed.

I assume they have different Kp, Ki, Kd values and maybe different thermistors. Can you have multiple bed configurations `[heater_bed]` (like you can have multiple extruders) in your `printer.cfg`?

(If you're interested: https://www.thingiverse.com/thing:4291299)

KevinOConnor, 2020-06-01:
There is no automated system to do this.  It is possible to create two different config files and swap them in when changing the hardware.

-Kevin

KevinOConnor, 2020-06-14:
I'm going to close this as it looks like the original question was answered (there is no automated solution but there are workarounds).

-Kevin

closed 2020-06-14 by KevinOConnor