Klipper3d/klipper #2927: Unknown command:"M106"

gtavo, 2020-05-27:
I was trying to do a PID tunning, and realized that when I send the M106 command, I get a "Unknown command:"M106"" M106 S0 is fine, M107 also, but if i try M106 S255 , M106 S1.0,  I get the unknown command...

When is printing it turns on fine on the 2nd layer (set in the slicer)...

[klippy.log](https://github.com/KevinOConnor/klipper/files/4692205/klippy.log)

jakep82, 2020-05-28:
You need to define a `[fan]` section to control a part cooling fan with M106.

gtavo, 2020-05-28:
> 
> 
> You need to define a `[fan]` section to control a part cooling fan with M106.

Solved, thanks! @jakep82 , my fan configuration now, I like to keep my printer without fans making noise while not printing, so I have the E3D V6 and the drivers cooling fan connected to the second extruder pin on my SKR 1.4

[fan] #layer fan
pin: P2.3

[heater_fan E3D_V6]  #This fan will turn on when the extruder is on, and turn off when its temp is under 50
pin: P2.4
heater: extruder
heater_temp: 50.0
fan_speed: 1.0

closed 2020-05-28 by gtavo