Klipper3d/klipper #2928: Replicape / BlTouch Issues - Update Instructions

chappo, 2020-05-28:
Hi Gents,

Apologies in advance! I know this has been a tricky subject. Thanks for all the recent works done on the replicape & getting BLTouch working. I have just tried to re-install using the Debian 9.9 IOT recommended but still unable to get the servo's moving. Using a genuine BlTouch 3.1.

What are the steps required to get this going? Can we get the documentation updated with steps required? Or an image that has everything working? 

I have updated the kernel to Linux beaglebone 4.14.108-ti-r135
Tried to use the universala cape after disabling wifi / hdmi. (I haven't disabled the emmc but no longer using that due to space limitations)
Tried using the universal cape
Tried using servo0 and servo1, using Bltouch in the config and also defining a servo

@jannau @jakep82 I can see you guys have this working - any assistance?

From issue 2514 that we are expecting to see a pwmchip4? I don't have that currently.

```
/sys/class/pwm/pwmchip1:
total 0
lrwxrwxrwx 1 root pwm    0 May 28 00:32 device -> ../../../48300200.pwm
--w--w---- 1 root pwm 4096 May 28 00:32 export
-r--r--r-- 1 root pwm 4096 May 28 00:31 npwm
drwxrwxr-x 2 root pwm    0 May 28 00:31 power
drwxrwxr-x 3 root pwm    0 May 28 00:31 pwm-1:0
drwxrwxr-x 3 root pwm    0 May 28 00:31 pwm-1:1
lrwxrwxrwx 1 root pwm    0 May 28 00:32 subsystem -> ../../../../../../../class/pwm
-rw-rw-r-- 1 root pwm 4096 May 28 00:31 uevent
--w--w---- 1 root pwm 4096 May 28 00:31 unexport

/sys/class/pwm/pwmchip3:
total 0
lrwxrwxrwx 1 root pwm    0 May 28 00:32 device -> ../../../48302200.pwm
--w--w---- 1 root pwm 4096 May 28 00:32 export
-r--r--r-- 1 root pwm 4096 May 28 00:31 npwm
drwxrwxr-x 2 root pwm    0 May 28 00:31 power
drwxrwxr-x 3 root pwm    0 May 28 00:31 pwm-3:0
drwxrwxr-x 3 root pwm    0 May 28 00:31 pwm-3:1
lrwxrwxrwx 1 root pwm    0 May 28 00:32 subsystem -> ../../../../../../../class/pwm
-rw-rw-r-- 1 root pwm 4096 May 28 00:31 uevent
--w--w---- 1 root pwm 4096 May 28 00:31 unexport

/sys/class/pwm/pwmchip5:
total 0
lrwxrwxrwx 1 root pwm    0 May 28 00:32 device -> ../../../48304100.ecap
--w--w---- 1 root pwm 4096 May 28 00:32 export
-r--r--r-- 1 root pwm 4096 May 28 00:31 npwm
drwxrwxr-x 2 root pwm    0 May 28 00:31 power
drwxrwxr-x 3 root pwm    0 May 28 00:31 pwm-5:0
lrwxrwxrwx 1 root pwm    0 May 28 00:32 subsystem -> ../../../../../../../class/pwm
-rw-rw-r-- 1 root pwm 4096 May 28 00:31 uevent
--w--w---- 1 root pwm 4096 May 28 00:31 unexport

/sys/class/pwm/pwmchip6:
total 0
lrwxrwxrwx 1 root pwm    0 May 28 00:32 device -> ../../../48304200.pwm
--w--w---- 1 root pwm 4096 May 28 00:32 export
-r--r--r-- 1 root pwm 4096 May 28 00:31 npwm
drwxrwxr-x 2 root pwm    0 May 28 00:31 power
drwxrwxr-x 3 root pwm    0 May 28 00:31 pwm-6:0
drwxrwxr-x 3 root pwm    0 May 28 00:31 pwm-6:1
lrwxrwxrwx 1 root pwm    0 May 28 00:32 subsystem -> ../../../../../../../class/pwm

```

[klippy.log](https://github.com/KevinOConnor/klipper/files/4692923/klippy.log)

KevinOConnor, 2020-07-10:
Alas, I don't have any way to assist with these types of Replicape Linux issues.  Perhaps @jannau , @profdryoman, or @goeland86 may be able to help?

-Kevin

jakep82, 2020-07-10:
@chappo I still have my Replicape, but I've moved it to a different printer and no longer own a BLTouch so I'm not able to offer any help here.

goeland86, 2020-07-10:
@chappo from what I recall, there needs to be a patch done to Klipper to use the proper PWM outputs - it's on my todo list of things to track down and document, and ultimately submit as a PR to @KevinOConnor. I don't have a BLTouch myself so it makes it a little difficult to trace, but I think the unofficial Klipper discord has some folks who may be able to help as well.

chappo, 2020-07-11:
Thanks Guys. Appreciate the work done to date. The Refactor

KevinOConnor, 2020-07-25:
Okay, it sounds like we can close this issue then.

-Kevin

closed 2020-07-25 by KevinOConnor