Klipper3d/klipper #2930: TL Smoothers

Hairyloon, 2020-05-28:
I happened upon this article: https://all3dp.com/2/tl-smoother-should-i-add-one-to-my-3d-printer/?omhide=true
I'd never heard of TL smoothers, but my drivers are of the type that this article recommends they are good for, but I wondered if it was a problem that Klipper dealt with in another way, or is it worth me adding some.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4695539/klippy.log)

joleeee, 2020-05-28:
I'm not sure if this is the place to ask about this stuff, you might want to try the discord server or subreddit. Either way they can improve print quality even on Klipper, yes.

Asterchades, 2020-05-29:
Also be aware that they don't work on more specialised drivers, like pretty much anything from TMC. They're best suited to more generic drivers that have been repurposed for 3D printing, such as the very common A4988 found on cheaper Chinese printers or the DRV8825.

Hairyloon, 2020-05-29:
> 
> 
> I'm not sure if this is the place to ask about this stuff...

Nor was I, hence my other post seeking guidance on posting.
Having received none, I figured this fits well into the "Try it & see" category.

> 
> 
> Also be aware that they don't help on more specialised drivers, like pretty much anything from TMC. They're best suited to more generic drivers that have been repurposed for 3D printing, such as the very common A4988 found on cheaper Chinese printers or the DRV8825.
> 

Mine has DRV8825, these smoothers appear to be no more than a handful of diodes, and I have literally hundreds of those to spare. I might as well throw a few together and give it a try.

Any suggestions on what's the best model to demonstrate any improvement?

jakep82, 2020-05-29:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#mailing-list

Hairyloon, 2020-05-29:
For what reason do you suggest a mailing list?
They weren't a great means of discussion back when they were the best thing we'd thought of.

Asterchades, 2020-05-30:
> Mine has DRV8825, these smoothers appear to be no more than a handful of diodes, and I have literally hundreds of those to spare. I might as well throw a few together and give it a try.

They look like that because that's exactly what they are. It's diodes, run opposite each other (to allow bi-directional flow), in series with each phase of the stepper. The variation between them (though it's harder to find different options any more) is simply the number of diodes - the most common have 2 diodes in series per direction per phase.

https://hackaday.com/2019/10/16/using-tl-smoothers-for-better-3d-prints/ goes into more detail, including a link to the original research by Schrodinger Z. Conveniently he's using DRV8825s for his analysis.

closed 2020-06-14 by KevinOConnor