Klipper3d/klipper #2931: Runout sensor - printer start printing without filament

Stetinac, 2020-05-28:
Hi,

printer start printing without detecting filament by runout sensor.
It's  possible to test status of runout sensor before printer start printing and sensor is enabled at cfg?

My printer setup is: Rpi4 and Octoprint

Command: "**QUERY_FILAMENT_SENSOR SENSOR=runout_sensor**"
returns: "**Filament Sensor runout_sensor: filament not detected**"

Best regards,

Pavel

[klippy.log](https://github.com/KevinOConnor/klipper/files/4698756/klippy.log)

Arksine, 2020-05-29:
Right, the purpose of the sensor is to detect a filament runout during a print, it would not disallow you from starting a print if no filament is in the printer.  You could use a gcode macro to accomplish something like that:

```
[gcode_macro PRINT_START_CHECK]
gcode:
  {% if not printer["filament_switch sensor runout_sensor"].filament_detected %}
  # Emergency shutdown to abort print
  M112
  {% endif %}
```

Where "my_sensor" is the

Stetinac, 2020-05-29:
Thanks for help.

closed 2020-05-29 by Stetinac