Klipper3d/klipper #2934: Flashing bricks my board.

iondrei, 2020-05-30:
Hi guys,

So I've been trying to get Klipper on my printer and I get stuck after flashing the printer board. I have a stock Trigorilla 1.1 on an Anycubic Kossel Linear Plus. Followed all the steps "by the book". I get no errors or fails in the actual flashing. Putty says the "all ok" but the printer display shows 
[image]

After this, octoprint can't connect anymore to the printer, it just times out. Tried AUTO and fixed options (usb0 and 250000) and no luck. 
Here's the log from the octoprint terminal.
[octolog.txt](https://github.com/KevinOConnor/klipper/files/4705648/octolog.txt)

I also tried some "makes" with the low-level configs and still no luck.
After the bricking, I can easily flash the marlin fw using Arduino...FYI. 

Any advice, or ideas? 

Here's also a photo of my board if it helps in any way. Maybe you see something I haven't.
[image]

Thanks

jalanjarosz, 2020-05-30:
@iondrei 
Using the right terms is important.   "Bricked", in your description is not correct.  If your board was truly "bricked", you would not be able to re-flash Klipper or Marlin on it.   You would need additional hardware in order to attempt to recover it.

What you have is an incorrect configuration.  Uploading the klipper.log file would be helpful.

From what you describe, you have the wring USB ID in your printer.cfg file.

1) When you FLASH the low-level MCU code on, you have an initial USB ID.
2) After you FLASH the MCU, and it reboots, you will have a Klipper named USB ID.  This is the ID that you need to put into your printer.cfg file.

Check into that and see if that's your proplem.

iondrei, 2020-05-30:
Hi,
Yhea...sorry. Forgot the log, you're right.
[klipperlog.txt](https://github.com/KevinOConnor/klipper/files/4706088/klipperlog.txt)

Sorry for the miss-use of the term. Should have used "". As I said it can be flashed back so it is clearly not bricked. Also, I'm sort of a beginner when it comes to coding so I can do basic stuff and follow instructions and understand how things work.
I hope you're also expecting non-coders to also try your software too. :)

Thanks and sorry for my ignorance.

jalanjarosz, 2020-05-30:
@iondrei 
Looks like your issues is that your are missing the printer.cfg file:  

`Unable to open config file /home/pi/printer.cfg`

This is the file that contains all the printer configuration items that is specific for your board and printer.

iondrei, 2020-05-30:
@jalanjarosz 
Maybe I missed something but following the installations steps here: https://www.klipper3d.org/Installation.html

I get stuck just after this step:

sudo service klipper stop
make flash FLASH_DEVICE=/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 (of course with my serial port name. This is just a copy/paste of the step from the installation instructions :) )
sudo service klipper start

When or where is the printer.cfg file created or uploaded?

jakep82, 2020-05-30:
The configuring Klipper section in that document explains how to find and copy the appropriate config file into your home directory. Klipper will not start without a properly configured printer.cdg file.

iondrei, 2020-05-30:
@jakep82 
Indeed it is very clear, I agree but it doesn't say when. If i take the order of things it's after the make flash command that which limits access to my printer. I'm not too advanced with either coding or what the correct order of the steps is in a situation like this. Anyways...i deduce from your answer that I need to upload the printer config and then flash the device again with the make flash FLASH_DEVICE command, or I can just upload it and reboot the printer and it would work. I can't try it right now because I'm away from the printer...justa asking for later.

Thanks btw to both @jalanjarosz and you for your answers till now. I'm getting there. slowly but I'm getting there. :)

jakep82, 2020-05-30:
You don't need to flash again if the first flash was successful.  Klipper is not like Marlin.  Configuration changes are made by editing the printer.cfg file and issuing a restart command from the Octoprint console.  You never need to reflash unless you update the firmware, and even then you often don't need to reflash.

iondrei, 2020-05-31:
Printer config file solved it.
Thank you!

closed 2020-06-14 by KevinOConnor