Klipper3d/klipper #2937: install-octopi.sh : KLIPPY_ARGS not updated (in /etc/default/klipper)

veevee2, 2020-05-31:
Hello,
I just discovered a configuration error when installing klipper two times from differents locations

1. I did the instruction for installing (git, etc...) BUT ... from /tmp (old safety practice)
2. i realized it's more appropriate to do the install from /home/pi as files have to stay here for long
3. then i deleted the tmp git files and redid the install from /home/pi

finally the service never started successfuly with no /tmp/klipper.log or /tmp/printer available
It was specificaly the KLIPPY_ARGS  in /etc/default/klipper which where still pointing to some /tmp/klipper/xxx paths

The reason seems to be that /etc/default/klipper is not updated at each install.

I fixed it by deleting /etc/defaut/klipper then redid the install, everything became normal then.

Wouldn't be more "rookie robust" to update this configuration file at each install ?

Thanks,
Olivier

veevee2, 2020-05-31:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4707989/klippy.log)
a useless log, but to comply with ticket rules

KevinOConnor, 2020-06-01:
The current behaviour is intentional as we don't want to automatically overwrite a config file that the user may have customized.

-Kevin

veevee2, 2020-06-02:
> The current behaviour is intentional as we don't want to automatically overwrite a config file that the user may have customized.
> 
> -Kevin

Ok i understand,
Maybe just a little verbose message to Tell that klipper as not been found in the place specifiwd by the configuration file ?
Olivier,

KevinOConnor, 2020-06-14:
I'm going to close this as it looks like the original question was answered (the current behaviour is intentional).

-Kevin

closed 2020-06-14 by KevinOConnor

veevee2, 2020-06-14:
Ok for me
Thanks