Klipper3d/klipper #2938: Klipper Shutdown during printing

Bald-79, 2020-05-31:
I use PrusaSlicer. Configured pressure_advance: 0.72. During printing, the Klipper Shutdown. On the attached photo, the place where it turns off, I assume that this is connected with the bridges. If pressure_advance is not used, then printing is normal. I also attach a copy from the terminal with the last accepted gcode command.
[image]
[Terminal.txt](https://github.com/KevinOConnor/klipper/files/4708241/Terminal.txt)

[klippy.log](https://github.com/KevinOConnor/klipper/files/4708232/klippy.log)

theopensourcerer, 2020-05-31:
Message in the log says "MCU 'mcu' shutdown: Stepper too far in past
This generally occurs when the micro-controller has been
requested to step at a rate higher than it is capable of
obtaining."

Your pressure advance setting seems very high to me. And your extruder step distance is very small.

Bald-79, 2020-06-03:
> Message in the log says "MCU 'mcu' shutdown: Stepper too far in past
> This generally occurs when the micro-controller has been
> requested to step at a rate higher than it is capable of
> obtaining."
> 
> Your pressure advance setting seems very high to me. And your extruder step distance is very small.

pressure_advance exactly as much as the test turned out.  Step so much is needed for the lv driver.  I changed the driver to tmc2209 and everything began to work.  but this is a bug.  why turn off the printer if the processor did not cope with the calculations, we must move on to the following, let the defect on the model be better than everything in the trash.

KevinOConnor, 2020-06-25:
I'm going to close this as it looks like a solution was found for the original issue.  There is a FAQ on error messages in Klipper: https://www.klipper3d.org/FAQ.html#why-does-klipper-report-errors-i-lost-my-print

-Kevin

closed 2020-06-25 by KevinOConnor