Klipper3d/klipper #2939: Can't access USB on MKS SGen-L

rbrian, 2020-05-31:
don't have a useful log at this point. if someone can give me steps to get a log that has something, i'll gladly try it out. 

Steps I took:

- run make menuconfig - pick 176x, 1768, and USB option. Uncheck bootloader bot. 
- make, and copy out/klipper.bin to PC. Rename to firmware.bin, and copy to board's SD card. 
- Reinstall SD card, and power on the system. 

In the process of creating my printer.cfg, tried to do "ls /dev/serial/by-id". got "folder not found". looking at /dev, there is no /serial folder. dmesg shows: (screenshot) https://www.screencast.com/t/LyeXH6QF4.

With assistance from some kindly folks in IRC, tried to get it talking via the board's serial UART. Unsuccessfully. 'status' gave either "trying to connect" or "connection unsuccessful". I'd rather go with USB anyway, to be honest.

BlackStump, 2020-06-01:
Have you tried the default settings for a LPC176x  ie as per pic?

[image]

rbrian, 2020-06-01:
my settings have been the same, except for the bootloader, which this board doesn't have. i'll try with that set and see if it makes a differenc.e

rbrian, 2020-06-01:
acquired a BTT SKR 1. 3 board, which is also a 1768-based board. Same issues with USB. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4713922/klippy.log)

rbrian, 2020-06-01:
BINGO! don't understand why, but selecting USB **AND** bootloader works. Will close issue after further testing.

rbrian, 2020-06-02:
USB seems reliable so far. Apparent non-issue. Although I wonder why it does it this way... can anyone comment on that?

KevinOConnor, 2020-06-05:
The board does have a bootloader and that option must be specified during "make menuconfig".

-Kevin

closed 2020-06-05 by rbrian