Klipper3d/klipper #2940: Klipper homing not respecting triggered endstops with Fysetc S6 and 2209

Ayasz-Art, 2020-06-01:
Just installed a new Fysetc S6 board with 2209. Got just about everything to work however when trying to home only the X axis endstop is recognised when triggered, and even then it rams into it for a good second before retracting, then rams into it again after. When homing from a triggered position it retracts then rams into it again for a few skips. The Y and Z endstops are completely ignore and will ram till safety timeout. All endstops are working with m119 and I have even tried replacing with a brand new endstop and wires, no change. Reflashed firmware several times testing different make options, only the 12mhz crystal option works. I have also tested sensorless just for the sake of it and that works. Tried cutting the diag pin, tried different jumper arrangments. Klipper will not respect when endstop status is changed to triggered, accept for that weird and delayed responsce to X.
[Klipper FYSETC S6 Backup.txt](https://github.com/KevinOConnor/klipper/files/4709059/Klipper.FYSETC.S6.Backup.txt)

BlackStump, 2020-06-01:
I would suggest checking your wiring to make sure the endstop's are wired to the correct axis on the mcu.

I would suggest following the Config-checks because trying to home with out checking that the endstop's actually work is not a good idea.
https://www.klipper3d.org/Config_checks.html

Ayaxoxo, 2020-06-01:
Thank you for looking into this. 

I have already stated above that all endstops are confirmed working with the m119 command. m119 = QUERY_ENDSTOPS. I have now redone all the tests using the "QUERY_ENDSTOPS" command to satisfy your suggestion exactly. Klipper reads endstops as triggered on the correct axis but does not respect. I have tried manually pressing and holding the endstops while the printer is given the home command and it simply ignores the triggered status.  

To be thorough I have tested all possible arrangments of triggeting endstops, 1 mat a time, two at a time, all 3. All return positive results on the correct axis when queried. But when homing, are being ignored.

BlackStump, 2020-06-01:
just for clarity the endstop reports open when not depressed and triggered when then endstop is depressed?

Ayaxoxo, 2020-06-01:
Yes, Open when not pressed, and triggered when pressed. And on the correct axis.

BlackStump, 2020-06-01:
sorry I am not familiar with that board so hopefully someone with experience with S6 will chime in.

KevinOConnor, 2020-06-01:
You'll need to attach the Klipper log file from the event in order to obtain assistance here - see: https://www.klipper3d.org/Contact.html

That said, I'd guess some kind of jumper setting on the board is not correct, or possibly the lack of pullups in the endstop_pin declaration.

-Kevin

Ayaxoxo, 2020-06-01:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4713750/klippy.log)
Thank you for checking. I had some folks over in the Klipper discord have a look at it and we went through a number of different pin arrangements, even cutting driver pins and destroying a driver entirely for the sake of the experiment. They ended up telling me that it sounds like some sort of a bug and to reach out to yourself. But there is also a good chance I missed something on the board somewhere despites checking over it multimple times, the documentation and support is limited. 

I saw there was a Klipper update yesterday and so I have reflashed the firmware just to be up to date but no change. Originally tried with the same version the old printer was running from about 2 months ago, also same results. 

Thank you

Ayaxoxo, 2020-06-01:
Problem is fixed now with the pullup declaration in the endstop_pin. I had 4 people look over the config file and noone picked up on this. Thank you very much and sorry to waste any time. I was genuinely stuck on this for days and many hours. 

Thank you again.

closed 2020-06-01 by Ayaxoxo