Klipper3d/klipper #2941: Feature Request: [fan] start fan early based on kick_start_time or new variable?

madsi1m, 2020-06-01:
Is it possible to have klipper look ahead of the gcode so that the fan is spinning at the correct speed before it reaches the point where the fan should be at the exact speed? Or have a new option to start he fan early by x seconds?

For example in my config i need a kick_start_time of 0.800. When i print a bridge the start of the bridge does not have enough air flowing to it. It would be better if the fan is already at speed just before the bridging starts.

Cheers

madsi1m, 2020-06-01:
Attached log [klippy.log](https://github.com/KevinOConnor/klipper/files/4709333/klippy.log)

KevinOConnor, 2020-06-01:
There is no current support for this.  It would require an interested developer to implement and submit.

-Kevin

madsi1m, 2020-06-02:
I noticed the docs talk about a look ahead used for cornering, could part of that be used?

madsi1m, 2020-06-03:
What do you think of this?

```
[gcode_macro M106]
rename_existing: M10106
default_parameter_S: 0
gcode:
    {% if S|int > 0 %}
      G4 P{ printer.configfile.config["fan"]["kick_start_time"]|float * 10000 }
    {% endif %}
    M10106 S{S}
```

KevinOConnor, 2020-06-25:
I'm going to close this as it looks like the conversation has concluded.  I'm not sure about your example macro - hopefully you've gotten a chance to test it yourself.

-Kevin

closed 2020-06-25 by KevinOConnor