Klipper3d/klipper #2942: Is it possible to still use a z endstop for if/when bltouch fails?

madsi1m, 2020-06-01:
Hi,

I use a SKR Mini E3 1.2 which has TMC2209 with sensorless homing enabled for X and Y. For my Z i have a bltouch.

I am wondering how best to or if it is possible to also use sensorless homing on the Z? For example: In the situation where my bltouch has failed and my nozzle hits the bed, I would like for the printer to stop or run some custom gcode (lift the z and stop everything).

I noticed under [display] there is a kill_pin but i can not connect it to `tmc2209_stepper_z:virtual_endstop` without generating errors.

Cheers

madsi1m, 2020-06-01:
Attached log [klippy.log](https://github.com/KevinOConnor/klipper/files/4709333/klippy.log)

KevinOConnor, 2020-06-01:
Currently, only one endstop pin can be used during homing and that pin is only ever used during homing (there is no runtime endstop checking).  It is possible to declare a `[gcode_button]` section to run a particular action on a switch event though.

-Kevin

madsi1m, 2020-06-01:
Thanks Kevin, I seem to have missed that when going through the example files.

Here is what i ended up doing:

```
[gcode_button kill_switch]
pin = PC14
press_gcode =
        M104 S0
        M140 S0
        M106 S0
        G91
        G1 Z20 F5000
        M84
        M112
```
This way when the nozzel smashes the bed it will go back up and hopefully reduce the damage

madsi1m, 2020-06-01:
would be very cool if i could use `pin: tmc2209_stepper_z:virtual_endstop` as `gcode_button` but i get this error:
```
AttributeError: TMCVirtualPinHelper instance has no attribute 'register_config_callback'
```

KevinOConnor, 2020-06-05:
As you've seen, the virtual_endstop pin is exclusively for use as an endstop.  Otherwise, it seems you've found a solution.

Cheers,
-Kevin

Hairyloon, 2020-06-08:
After my probe failed causing the head to ram the bed, I wired the probe in parallel with the microswitch, on the grounds that the probe should trigger first, and the switch would be a backup if it went too far.
I forget now why it didn't work...

madsi1m, 2020-06-09:
I prefer to keep them separate, so i can stop everything on a fail event

KevinOConnor, 2020-06-25:
I'm going to close this as it appears the original question has been answered.

-Kevin

closed 2020-06-25 by KevinOConnor

marcosscriven, 2020-11-10:
> This way when the nozzel smashes the bed it will go back up and hopefully reduce the damage

@madsi1m - does the gcode actually stop the z-move immediately?

KevinOConnor, 2020-11-10:
If the gcode_button uses `{action_emergency_stop(msg)}` (see docs/Command_Templates.md for the details) then the shutdown will occur as soon as the host gets notified of the button press.  (It will not wait for existing g-code moves to complete.)

-Kevin