Klipper3d/klipper #2943: Klipper: G2/G3 only supports relative extrude mode

Andrei-Pozolotin, 2020-06-01:
please confirm if this is a klipper-specific issue (and not octoprint plugin issue)
https://github.com/FormerLurker/ArcWelderPlugin/issues/23

KevinOConnor, 2020-06-01:
As the error indicates, Klipper currently only supports G2/G3 commands in relative extrusion mode.  It's possible to implement support for absolute extrusion mode, but that would require an interested developer to implement and submit it.

-Kevin

FormerLurker, 2020-06-02:
I may give it a go.  Will check out the code and see if I can figure it out.  I think g2/g3 code is going to become more common after arc welder comes out of beta, so might as well get a jump on things.

Andrei-Pozolotin, 2020-06-02:
please keep in the loop: will gladly test that

KevinOConnor, 2020-06-04:
FYI, I took a look at the previous commit that disabled absolute extrude moves and I don't think it was needed.  So, as of commit 5d26c2a0 I think absolute extrude moves for G2/G3 should work.

-Kevin

Andrei-Pozolotin, 2020-06-04:
thank you. will test and report

closed 2020-06-08 by klipper-gitissuebot