Klipper3d/klipper #2944: FR: M108 as simple break from M109/M190

Andrei-Pozolotin, 2020-06-01:
1. this [how-do-i-cancel-an-m109m190-wait-for-temperature-request](https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-cancel-an-m109m190-wait-for-temperature-request)

2. feels like a major overkill (total reset of all things)
```
Navigate to the OctoPrint terminal tab and issue an M112 command in the terminal box. The M112 command will cause Klipper to enter into a "shutdown" state, and it will cause OctoPrint to disconnect from Klipper. Navigate to the OctoPrint connection area and click on "Connect" to cause OctoPrint to reconnect. Navigate back to the terminal tab and issue a FIRMWARE_RESTART command to clear the Klipper error state. After completing this sequence, the previous heating request will be canceled and a new print may be started.
```
3. how hard would it be to implement "marlin behavior"?
https://marlinfw.org/docs/gcode/M108.html
which merely breaks out of M109/M190 wait loop upon M108
so both host and mcu keep their state

KevinOConnor, 2020-06-05:
I think the FAQ answers this as best we can.  The M108 command is broken (as it does not cancel commands sent after M109/M190) and I, personally, am not interested in implementing or maintaining it.

-Kevin

Andrei-Pozolotin, 2020-06-06:
1. ok, clear.

2. still, how would you implement proper `BREAK` command
which does not require `OctoPrint` disconnect and `MCU` reset?

KevinOConnor, 2020-06-06:
You can look at how M119 is handled in klippy/gcode.py .

FWIW, though, I'd say the best way to cancel a heating operation would be to implement it via a mechanism other than g-code.  There's been discussion on adding a "web interface" to klipper which could potentially improve this type of user interaction.  There's no timeframe for that though.

-Kevin

KevinOConnor, 2020-06-25:
I'm going to close this as it looks like the conversation has concluded.

-Kevin

closed 2020-06-25 by KevinOConnor

charlespick, 2021-04-28:
I am left a little confused, I use M108 all the time to stop a print before it's started and find it a very useful feature on Marlin when combined with emergency parser so I don't know what you mean by "The M108 command is broken." Could you please explain? 

If I were to implement it myself would it likely be approved to merge in?