Klipper3d/klipper #2946: i need a bit of orientation

rmalchow, 2020-06-02:
hi,

i have a question that isnt really related to 3d printing. i am using leadscrews and steppers to position a camera along two axis, and i need to be able to tell it to go to a position, then let me know it arrived (another thing would be that to take up the slop in the system, it would be nice to always approach the final position from the same direction). 

i have various options of implementing this - i have used the raspberry stepper hat from waveshare, i have also used arduinos. the downside of these is that i have to deal with implementing the low level stepper motor controls myself. i did this on a very basic level, and it is working. however, i am looking to improve speed and acceleration curves to get a faster (and possibly also smoother) ride.

so i am looking into options, and klipper seems to be one of them. however, i am confused. maybe this is all stuff that is completely clear to people who are dealing with this more, but my main problem is that i am trying to understand what the components are (this is reasonably clear), what their responsibilities are (already getting a bit murky) and what the communication protocols are between them (i have absolutely zero clue).

my current understanding looks like this (and this might be seriously wrong):

[image]

is there anyone who can describe in a little bit more detail what's going on? if i know a bit more, i'd be happy to write it up for the klipper documentation if you want me to.

toodles

.rm

Hairyloon, 2020-06-02:
Sounds interesting.
Apart from to say that the communication protocol between octoprint and klipper is simply G-code, which I assume you already knew, I haven't a clue what the answers are, but I would be interested to find out.

KevinOConnor, 2020-06-05:
See https://www.klipper3d.org/Overview.html#developer-documentation for information on how the code operates.

-Kevin

KevinOConnor, 2020-06-25:
I'm going to close this as it appears the conversation has concluded.

-Kevin

closed 2020-06-25 by KevinOConnor