Klipper3d/klipper #2947: Problem with Bed Leveling and Fading

SeiteBroken, 2020-06-02:
So I have this issue with the Fade_start. I don't want to use fading but if i but it out of the config file it goes back to its default value. And the problem is that I have overextrusion on the hight of fade_start.
The overextrusion stops shortly after, but I can't get it totally out of the print. When fade_start=0.0
I have an elephant foot on my print.

On the left fade_start=0.0 & on the right fade_start=1.0

[image]

BlackStump, 2020-06-02:
It is covered in the Doc's [here](https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Mesh.md)

Difficult to give help when the required **Log** in not supplied

SeiteBroken, 2020-06-02:
So there is the log file, but I in the Docs i didn't found the isse or the solution. I tried every combination of fade_start and fade_end. So normally it should be deactivated.But the Issue remains

[klippy.log](https://github.com/KevinOConnor/klipper/files/4717692/klippy.log)

Arksine, 2020-06-07:
As the documentation indicates, if both `fade_start` and `fade_end` are left out of the configuration (ie set to their default values) OR if `fade_end` is less than `fade_start` then fade will be disabled.   If you experience issues with elephants foot or overextrusion when fade is disabled then it isn't related to fade.

KevinOConnor, 2020-06-25:
I'm going to close this as it appears the original question was answered.

-Kevin

closed 2020-06-25 by KevinOConnor