Klipper3d/klipper #2949: SKR Pro 1.1 HID Bootloader

schnello, 2020-06-03:
Hello.

Can anybody confirm that the new hid Bootloader for the SKR Pro works? Anytime if i flash the https://github.com/Arksine/STM32_HID_Bootloader/releases/download/v0.6-beta/hid_bootloader_SKR_PRO.bin with ST Link any further USB connection is not possible anymore (no USB device, no /dev/ttyACM0 etc etc)

Arksine, 2020-06-07:
It works.   I did upload a new version built with an older toolchain (no code changes) as it seems that there were some mixed results with the version built against gcc 8.  I also enabled issue reporting on that repo so further issues can be reported there.

One thing to keep in mind is that after flashing the bootloader you must manually enter it by setting BOOT1 high with a jumper.  After Klipper is flashed this is no longer necessary as "make flash" will enter the bootloader automatically and then upload Klipper.

schnello, 2020-06-07:
Closed and open https://github.com/Arksine/STM32_HID_Bootloader/issues/1

closed 2020-06-07 by schnello