Klipper3d/klipper #2952: RepRapDiscount Full Graphic Smart Controller on SKR 1.4 pinout

Ducati87, 2020-06-04:
Hello, I'm trying to setup the RepRapDiscount Full Graphic Smart Controller (LCD 12864) on my SKR 1.4 Turbo but I cannot find any pinout configuration that could help me. Already tried a few configuration but without good results. May anyone help me to configure it? Thank you.

schnello, 2020-06-05:
In the SKR1.4 Config:
https://github.com/KevinOConnor/klipper/blob/5d26c2a08ec0dafcacbf47de4d9ea619a324027a/config/generic-bigtreetech-skr-v1.4.cfg#L181-L195

From the LCD Config:
https://github.com/KevinOConnor/klipper/blob/5d26c2a08ec0dafcacbf47de4d9ea619a324027a/config/sample-lcd.cfg#L13-L26

Ducati87, 2020-06-05:
it works! Thank you so much!

closed 2020-06-05 by Ducati87

bdolph, 2020-12-16:
Ducati87 please send me the binfile for the btt skr V1.4 Turbo that you use to put on lcd. I don't get this right let me see if my lcd is working please : salvage1009@gmail.com

ichzwei, 2021-03-13:
I have also skr 1.4 turbo and a RepRapDiscount Full Graphic Smart Controller.
but i haven't got it to show anything on the display.

niolang, 2021-04-14:
> 
> 
> I have also skr 1.4 turbo and a RepRapDiscount Full Graphic Smart Controller.
> but i haven't got it to show anything on the display.

Are you using the LCD 12864 or LCD 2004?
Keep the board pinout as given by the klipper/config/generic-bigtreetech-skr-v1.4.cfg
https://github.com/KevinOConnor/klipper/blob/5d26c2a08ec0dafcacbf47de4d9ea619a324027a/config/generic-bigtreetech-skr-v1.4.cfg#L181-L195

bus use display hd44780 from  klipper/config/sample-lcd.cfg
https://github.com/KevinOConnor/klipper/blob/5d26c2a08ec0dafcacbf47de4d9ea619a324027a/config/sample-lcd.cfg#L29-L46