Klipper3d/klipper #2953: Pressure advance: is it material specific?

Hairyloon, 2020-06-05:
I'm still a beginner at this game. I'm successfully printing various things, but I believe there is plenty of scope to tune my printer to make it print better and faster.

I've tried setting the pressure advance, as per the instructions, I don't know if I got it right or not, but after I changed the filament I printed a plain box which I have printed several times before and it came out with similar artefacts to those exhibited on the test print for the pressure advance, where I had not noticed them on the same box in the previous filament. 
Specifically I refer to wibbly corners: wibbling this way on that corner and that way on this corner

It may be that I was looking at the wrong thing when I set the pressure advance and maybe it's something else altogether, but either way I figured it was worth an ask...
I can post a picture if it would help, but not right now as I don't have a good enough camera to hand.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4738863/klippy.log)

Asterchades, 2020-06-08:
It's very much per filament. I've found most filaments of the same type from the same manufacturer are close enough that you don't really need to tweak it, but certainly if you're changing material type (or adding a suspension) or manufacturer than you'll definitely want to re-tune to get a new baseline. That's not to say there won't be roll-to-roll variance, or influence by the pigment, just that it usually isn't significant in this case.

I know I run 0.1 for 3D Fillies PLA+ and PLASteel (PLA with a metal suspension), but closer to 0.2 for their PETG and somewhere in the middle for their PLASilk (PLA with a polymer blend added to make it shiney). These latter two are far less viscous when heated, though, so it makes sense that the value will be higher.

Note also that if you change temperatures it can be worth tuning again. Lower temperatures will require less advance due to the higher viscosity, and vice versa.

Hairyloon, 2020-06-09:
Thank you.
Are there any other suggestions as to a tuning model, other than the official cube in the instructions, which seems a bit excessive?
What prompted me to ask was a simple box printed in vase mode. I suppose I could just print those, at least until I run out of uses for them.

If temperature makes a significant difference, would it be worth adjusting the code to take account of that?
I'm not a coder, I don't know how hard it is to do.

Asterchades, 2020-06-09:
Well, as much as I trust these guys to know how their system works, I found it easier to get a good starting point with a kind-of D shape - about 4cm up the straight line (X), and 2cm perpendicular (Y), with a 5cm depth (Z). This gives me two square corners and two rounded ones to evaluate. 3 bottom layers for adhesion, 2 walls to ensure good extrusion, and the seam towards one of the rounded corners so I can get a view of it as well. When printing it I disable retraction just for extra torture, and run it with 0% infill and 0 top layers since they're superfluous.

It's not a particularly useful or decorative object (though I've used more than one as little desk storage containers now), but it's quick, uses little material, and is reasonably accurate in my experience. Plus it was easy to create the model in the first place.

There doesn't really need to be any code adjustment for temperature. It's just one variable of the equation which you should sort out before trying to dial in your Pressure Advance, much like you would get an idea of temperature before sorting out retraction.

georgebashi, 2020-06-09:
For what it's worth, I have found that the cube in the docs does a really good job of exposing PA issues. When printing it, I often have outer walls that look OK through a wide range of values, but the notch down the inner wall shows up blobby corners / under extrusion that helps me narrow down the exact value to use.

Hywelmartin, 2020-06-09:
for a lot of different angles 
https://www.myminifactory.com/object/3d-print-pa-test-103970
https://youtu.be/Su3vA7i4iTE

KevinOConnor, 2020-06-25:
I'm going to close this as it appears the original question was answered (yes, it is material specific).

-Kevin

closed 2020-06-25 by KevinOConnor