Klipper3d/klipper #2954: Constant errors with reccomended DWC2

mejobloggs, 2020-06-06:
Apologies for this question. I know it isn't directly Klipper but it's making my Klipper experience very frustrating and would love to resolve it.

Printing is fine, but many buttons etc on the DWC2 UI give me errors causing the UI to be useless outside of straight printing. (e.g uploading/starting a print works beautifully)

When I try to heat the bed or nozzle through DWC2 I get:

 - "G10 P0 S120 Failed - Reason: 500 Internal Server Error"
 - "M140 P0 S40 Failed - Reason: 500 Internal Server Error"

Many commands give me this error:
 - Error:  'dict' object has no attribute 'get_command'

E.g when I edit/save config.g, it asks to restart, I click Yes, and it gives me the above error

Thanks if anyone can help

BlackStump, 2020-06-07:
Duet Web Control  is developed for and is only supported for the Duet3D Controllers.
While a Translator for Klipper has been made elsewhere. DWC2 was never made for Klipper nor is it supported by Klipper. 
People that use DWC2 must understand the above and the caveats with using a webui that was never intended for Klipper use.

I would suggest you use Octoprint.

There is a webui in the wind that is designed for Klipper it is alpha atm and who know when or if it will be merged into klipper.

Just to reiterate The developer of DWC2 will not provide support for klipper.
This is not the right place for DWC2 support.

[dwc2 to Klipper translator](https://github.com/Stephan3/dwc2-for-klipper)
[try here for working dwc2 for klipper](https://github.com/pluuuk/dwc2-for-klipper)
[dwc2 developer](https://github.com/Duet3D/DuetWebControl)

mejobloggs, 2020-06-11:
Thanks especially for the second link. And very interesting to hear about Mainsail. I appreciate your help even though it wasn't the right place for me to ask.

I see a few people are working on a pull-request to keep the dwc2/klipper translator up to date:

https://github.com/Stephan3/dwc2-for-klipper/pull/75

Thanks again :)

closed 2020-06-11 by mejobloggs