Klipper3d/klipper #2959: PID_CALIBRATE error

yarus23, 2020-06-07:
`Internal error on command:"PID_CALIBRATE"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 291, in _process_commands
    handler(gcmd)
[klippy.log](https://github.com/KevinOConnor/klipper/files/4742972/klippy.log)

  File "/home/pi/klipper/klippy/gcode.py", line 142, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/pid_calibrate.py", line 38, in cmd_PID_CALIBRATE
    Kp, Ki, Kd = calibrate.calc_final_pid()
  File "/home/pi/klipper/klippy/extras/pid_calibrate.py", line 131, in calc_final_pid
    midpoint_pos = sorted(cycle_times)[len(cycle_times)/2][1]
IndexError: list index out of range
Internal error on command:"PID_CALIBRATE"
`

for 100 degrees it works for 200 it throws this error.

KevinOConnor, 2020-06-14:
You should have gotten a message stating:
```
Heater extruder not heating at expected rate
Transition to shutdown state: Heater extruder not heating at expected rate
See the 'verify_heater' section in config/example-extras.cfg
for the parameters that control this check.
```

That is the root cause of your issue.  I don't know of any mainstream printers that can successfully heat the bed to 200 degrees.

-Kevin

KevinOConnor, 2020-06-25:
I'm going to close this as it appears the original question was answered.

-Kevin

closed 2020-06-25 by KevinOConnor