Klipper3d/klipper #2960: Print work slow - Anet a8 Klipper

assafm25, 2020-06-08:
Hi, 
I use klipper for about 8 month.
All work well until the last 3 month.
I not use klipper and print for long time and when i try use it i see its work much slower then before (110)

I format my SD and re install and configure all but no news on the speed side.
`

[klippy.zip](https://github.com/KevinOConnor/klipper/files/4744764/klippy.zip)

`

assafm25, 2020-06-11:
Any one ?!

lbibass, 2020-06-11:
We need more information. What do you mean slow? Unresponsive? Slow print speeds? Stuttering?

assafm25, 2020-06-11:
Hi,

Slow print speeds
its fills like its print on standard speed (50 mm)

BlackStump, 2020-06-12:
Please supply the gcode that is printing slow for you.
Klipper is not slow so more then likely it is your slicer.

assafm25, 2020-06-13:
Hi,

All of my prints are slow and not feels like on 110.
i can add some gcodes but all the same.
i use CURA

Some of my last prints:
https://www.thingiverse.com/thing:3008262

http://www.thingiverse.com/thing:1852762

assafm25, 2020-06-14:
Mybe its a problem with my pi?

I Really don't know how to solved this.

KevinOConnor, 2020-06-25:
I'm not sure there is much we can do to assist here - Klipper moves the toolhead at the speed specified in the gcode file.  If it's not running as fast as you'd like, you should configure the speed in your slicer.

-Kevin

closed 2020-07-10 by KevinOConnor