Klipper3d/klipper #2961: Bed mesh resolution.

Hairyloon, 2020-06-08:
Is there any downside to having a lot of probe points when calibrating the bed mesh, other than the time it takes?
(And perhaps the inference that if you need to take so many, then your bed is probably fit for scrap.)

If it is just time, then is it possible to have two degrees of probing?
I'm thinking to create a detailed map of the bed surface would take a long time, but having done that, a subsequent probing of fewer points could check the bed hasn't moved any.

Which brings me to a secondary question: how often do people typically run a bed mesh calibration?
I do it any time I've moved the printer, or had any kind of incident that might disturb things, or if the printing starts going wrong.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4745843/klippy.log)

Arksine, 2020-06-10:
1) There is no downside to a high density grid, however there are diminishing returns.  

2) Not currently.  Some attempts to implement "tilt" functionality have been made but thus far no one has submitted a solution that has met the standard for merging.

3)  I run it before each print on my printers.

KevinOConnor, 2020-06-25:
I'm going to close this as it appears the original question was answered.

-Kevin

closed 2020-06-25 by KevinOConnor