Klipper3d/klipper #2962: G1 Command Problem

Bernouillle, 2020-06-08:
Hello everyone,
i have a problem with the G1 command. I get the answer that i have some problems in stepcompress.
Do you have a clue where the problem could be? See the attached klippy.log. Thanks a lot in advance!
[klippy.log](https://github.com/KevinOConnor/klipper/files/4747140/klippy.log)

Hywelmartin, 2020-06-08:
do you have a picture of the machine??

 #2793 ??

Bernouillle, 2020-06-08:
I am sorry, but i can't atm. I developed a new kinematics class for my masters thesis and i control the axes with six motors in total and i use a Rumba board.

Bernouillle, 2020-06-08:
> #2793 ??

I am not using an extruder atm,  just implementing the kinematics class.

Bernouillle, 2020-06-08:
I also had to add there new params to the G1 command. They are needed for each calculation of the stepper positions. When i split up one G1 command in two lines the printer can move the steppers according to the calculations, but the calculations are not correct, because i had to split up the G1 command line.
I don't think that the mcu is too weak to perform and need to work at a lower rate, because the printer can execute the calculations of the two separated lines.

Is it possible, that there is a cap for a line input?

Bernouillle, 2020-06-14:
@Hywelmartin i took a closer look at the change and commits that have been made since my last version. I upgraded my firmware and it looks promissing. The debug modus doesn´t return any errors now. Can test it in a week, but i´m very optimistic.

KevinOConnor, 2020-06-14:
If you've changed the code and it no longer works, then you'll have to track down why.

A "stepcompress o=5 i=0 c=323 a=0: Invalid sequence" error generally occurs when the low-level kinematic functions generate an "instantaneous movement" - that is a movement with infinitive velocity.  This is usually a logic error in the code.

-Kevin

KevinOConnor, 2020-06-25:
I'm going to close this as the original question was answered (at least as best as we could).

-Kevin

closed 2020-06-25 by KevinOConnor