Klipper3d/klipper #2965: Klipper does not connect to Arduino DUE

mikkhalich, 2020-06-09:
Good afternoon, installed Klipper on Orange pi pc + (armbian buster). Arduino DUE + ramps 1.5 is connected to Orange pi.

I can not get Klipper to connect to Arduino DUE
This is the octopi log:

```
Klipper state: Disconnect
Printer is not ready
The klippy host software is attempting to connect. Please
retry in a few moments.
Klipper state: Not ready
13:58:53
Printer is not ready
```

The firmware of the device goes fine:

```octoprint@orangepipcplus:~/klipper$ make flash FLASH_DEVICE=/dev/serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Due_Prog._Port_75834303438351100151-if00
  Flashing out/klipper.bin to /dev/serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Due_Prog._Port_75834303438351100151-if00
Entering bootloader on /dev/serial/by-path/platform-1c1c400.usb-usb-0:1:1.0
Device reconnect on /dev/serial/by-path/platform-1c1c400.usb-usb-0:1:1.0
lib/bossac/bin/bossac -U -p /dev/serial/by-path/platform-1c1c400.usb-usb-0:1:1.0 -e -b -w out/klipper.bin -v

Erase flash

Done in 0.058 seconds
Write 20508 bytes to flash (81 pages)
[==============================] 100% (81/81 pages)
Done in 4.207 seconds
Verify 20508 bytes of flash
[==============================] 100% (81/81 pages)
Verify successful
Done in 4.030 seconds
Set boot flash true
octoprint@orangepipcplus:~/klipper$
```

printer.cfg - MCU part :

```
[mcu]
serial: /dev/serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Due_Prog._Port_75834303438351100151-if00
#serial: /dev/ttyACM0
#   The serial port to connect to the MCU. If unsure (or if it
#   changes) see the "Where's my serial port?" section of the FAQ. The
#   default is /dev/ttyS0
#baud: 250000
#   The baud rate to use. The default is 250000.
pin_map: arduino
#   This option may be used to enable Arduino pin name aliases. The
#   default is to not enable the aliases.
#restart_method:
#   This controls the mechanism the host will use to reset the
#   micro-controller. The choices are 'arduino', 'cheetah', 'rpi_usb',
#   and 'command'. The 'arduino' method (toggle DTR) is common on
#   Arduino boards and clones. The 'cheetah' method is a special
#   method needed for some Fysetc Cheetah boards. The 'rpi_usb' method
#   is useful on Raspberry Pi boards with micro-controllers powered
#   over USB - it briefly disables power to all USB ports to
#   accomplish a micro-controller reset. The 'command' method involves
#   sending a Klipper command to the micro-controller so that it can
#   reset itself. The default is 'arduino' if the micro-controller
#   communicates over a serial port, 'command' otherwise.

```
What am I doing wrong?

[printer.zip](https://github.com/KevinOConnor/klipper/files/4750419/printer.zip)
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4750422/klippy.zip)

FHeilmann, 2020-06-09:
Which port on the due are you using? If I understand correctly the due has 2 USB ports, one for flashing and one for "other use". You should only need the programming port for the initial flash, after that you should use the other USB port.

jakep82, 2020-06-09:
Also note that if you're using a clone Due, most of them won't run properly unless you press the reset button after power on.  They start in the bootloader instead of running the installed firmware.  I ended up buying a genuine Due which doesn't have this problem, but you can also modify clones to make them work.  See [this link](https://forum.arduino.cc/index.php?topic=256771.msg2512504#msg2512504) for more info.

mikkhalich, 2020-06-10:
> Which port on the due are you using? If I understand correctly the due has 2 USB ports, one for flashing and one for "other use". You should only need the programming port for the initial flash, after that you should use the other USB port.

thanks helped

> Also note that if you're using a clone Due, most of them won't run properly unless you press the reset button after power on. They start in the bootloader instead of running the installed firmware. I had this problem with a clone and got tired of pressing the reset button every time I turned on the printer, so I ended up buying a genuine Due which doesn't have this problem. You can also modify clones to make them work. See [this link](https://forum.arduino.cc/index.php?topic=256771.msg2512504#msg2512504) for more info.

thanks, i fixed my arduino due

closed 2020-06-10 by mikkhalich