Klipper3d/klipper #2967: Ender 3 - BLTouch as endstop - Nozzle crashes into bed then stops

federov31, 2020-06-09:
Hello, I'm struggling to configure my BLTouch as a Z_endstop on my printer (Ender 3 Pro - SKR Mini v1.2 - TriangleLabs BLTouch)

After following guide on Klipper Github, the BLTouch is working, pin is deploying but everytime I home the Z, the nozzle crashes into bed the stops then it raises 10mm above the bed ([safe_z_home] , home_xy_position)

I managed to set the BLTouch X and Y (X -40.7, Y -10.3) according to the procedure but I'm not sure about my Z offset [https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#calibrating-probe-x-and-y-offsets](Probe_Calibrate.md#calibrating-probe-x-and-y-offsets)

This is the mount I use for the BLTouch : [Ender 3 BL-Touch Mount
](https://www.thingiverse.com/thing:3091052)

May someone help me to configure it ?

```
[stepper_x]
step_pin: PB13
dir_pin: !PB12
enable_pin: !PB14
step_distance: .0125
endstop_pin: ^PC0
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_x]
uart_pin: PB15
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 250

[stepper_y]
step_pin: PB10
dir_pin: !PB2
enable_pin: !PB11
step_distance: .0125
endstop_pin: ^PC1
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_y]
uart_pin: PC6
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 250

[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
position_endstop: 0
position_min: -4.750
position_max: 250

[tmc2209 stepper_z]
uart_pin: PC10
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 5

[bltouch]
sensor_pin: ^PC2
control_pin: PA1
x_offset: -40.7
y_offset: -10.3
z_offset: -4.750

[safe_z_home]
home_xy_position: 117,117
speed: 50
z_hop: 10
z_hop_speed: 5

[extruder]
step_pin: PB3
dir_pin: !PB4
enable_pin: !PD2
step_distance: 0.010526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA0
control: pid
# Tuned for Ender 3 Stock hardware 210°C
pid_kp: 21.868
pid_ki: 0.884
pid_kd: 135.305
min_temp: 0
max_temp: 250

[tmc2209 extruder]
uart_pin: PC11
microsteps: 16
run_current: 0.650
hold_current: 0.500
stealthchop_threshold: 5

[heater_bed]
heater_pin: PC9
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC3
control: pid
# Tuned for Ender 3 Glass Bed 60°C
pid_kp: 75.154
pid_ki: 1.704
pid_kd: 828.571
min_temp: 0
max_temp: 130

[fan]
pin: PA8

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f103xe_34FFD8055055323913850143-if00

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[static_digital_output usb_pullup_enable]
pins: !PC13

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PB6, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB7, EXP1_10=<5V>

[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2
menu_timeout: 10

```

holstvoogd, 2020-06-09:
Hi, pretty sure the z_offset should be positive! It is inversed from marlin

BlackStump, 2020-06-09:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L558

jakep82, 2020-06-09:
Your probe is way too high.  It's not triggering until you're entire X axis flexes far enough to trigger the bltouch.  You need to move the bltouch down several millimeters.  You might be able to make your existing mount work using washers to space it down.