Klipper3d/klipper #2974: How to limit logfile size?

ZhenyaKa, 2020-06-10:
Is any options for that?

I have 2 deltas, /tmp placed at ramdisk.

I have 2 files and them oversized very quckly

Is any method to limit 5-6 print session in log file? And when create new log

[klippy.log](https://github.com/KevinOConnor/klipper/files/4758256/klippy.log)

Maybe anybody have well working logrotate config for that?

KevinOConnor, 2020-06-14:
There is no current mechanism to limit log file size.  However, the logs are automatically rolled over every day and only 5 days are kept.

-Kevin

ZhenyaKa, 2020-06-14:
Yes, but I have two printers connected to Orange Pi Lite and when they print both I got huge logs for my ramdisk (/tmp) :(

KevinOConnor, 2020-06-25:
I'm going to close this as the original question was answered (it is not currently possible).

-Kevin

closed 2020-06-25 by KevinOConnor