Klipper3d/klipper #2977: Add support for HX711 based ABL sensor (used in CR-6 printers)

jschuh, 2020-06-11:
The HX711 load cell ADC is used as a bed level sensor on the upcoming [Creality CR-6](https://www.creality.com/cr-6-se-3d-printer-back-us-to-save-100-p00296p1.html) series of printers. It works by treating the extruder mounting bracket as a load cell, allowing it to detect the load generated by the nozzle as it touches the bed surface.

I found [an old, bitrotted PR](https://github.com/MarlinFirmware/Marlin/pull/2309) to implement this in Marlin, which shows that the code itself is actually quite straightforward. The patch also references [a blog post (in Russian)](https://habr.com/en/post/378637/) that provides a full demonstration using a stock 5kg load sensor (rather than a custom bracket), meaning there's the potential for broader use beyond just the CR-6 printers.

I picked up 1kg and 5kg load cells to experiment with this, so if there's interest I could take a crack at adding the support sometime in the next month or so.

FHeilmann, 2020-06-11:
The immediate issue I'm seing with this is that the HX711 does not have an "interrupt pin" which allows to have a digital pin indicating a threshhold has been crossed. This is problematic because the HX711's current load has to be read by shifting in a 24-bit value, which is quite expensive compared to the simple "read digital pin" which klipper currently does.

These changes would have to be implemented in the low level MCU code as well, which is a bit trickier than writing the python host code.

If you're willing to tinker a bit, I have previously written this sketch for an attiny85 (a small arduino mcu), which connects to a HX711 and provides a digital output, similar to an endstop.

https://gist.github.com/FHeilmann/1e5e2414000afc2af3435a7804f2a84e

Hope that helps,

Florian

Hywelmartin, 2020-06-11:
@FHeilmann  do You know if it would be the same challenge to implement Precision piezo Orions analog out??? 
https://www.precisionpiezo.co.uk/product-page/precision-piezo-orion-kit

jschuh, 2020-06-12:
@FHeilmann, thanks for the tip. Fortunately, I have a lot of experience with low-level C, but just to make sure I get what you're suggesting:

The MCU code:
- Will need to get initialized from the host with a threshold for being triggered (and probably some other config parameters like number of samples, etc.).
- During probing will use a periodic timer to check when the sensor reading exceeds the threshold
- When the threshold is exceeded the code must trigger a "virtual interrupt."
- Probably needs some sort of debug command interface for the host (e.g. to read current sensor value, reset, etc.).
- _Question: The two wire protocol looks a lot like I2C, so do you think I can use the existing I2C MCU implementation to talk to the HX711?_

The host code:
- Looks like I can easily enough integrate this with the existing probe code on the host side (although I admit I haven't looked at this too closely yet).
- Any gotchas I might be missing here?

FHeilmann, 2020-06-12:
You'll essentially have to replicate what endstop.c does already. Just with the hx711 communication in place of the "gpio in read". I looked at it at some point but didn't follow through. I guess you'll end up with a lot of code duplication unless you've try to somehow make the endstop code generic enough so you can plug in digital pins and other systems.
@Hywelmartin doesn't the precision piezo stuff have digital outputs as well. I know my old precision piezo board does.

Hywelmartin, 2020-06-12:
@FHeilmann yes it does... But it would be nice to tune it from the terminal instead of the trimpot...

bung69, 2020-06-14:
The ability to monitor load cells from Klipper could also be useful to monitor how hard an extruder is pushing.  This could be used to dynamically slow down feed rate, or pause prints and alert the user. 

I also wonder if any improvements could be made to how pressure advance works, either actively with a closed loop, or through manual monitoring of a load cell followed by calibration of extra parameters relating to the non linear portion of how pressure is released over time. 

i have recently done some tests with a load cell attached in line with the extuder to measure different hot ends flow rate vs extruder force and was surprised at the detail visible on a Scope,  using a load cell and INA125 amplifier.  ( circuit from http://airtripper.com/1626/arduino-load-cell-circuit-sketch-for-calibration-test/ )

Maybe an analogue input to Klipper could be used if the poling rate was increased? 

or there is also a sparkfun Load cell I2C board as an alternative to the HX711.

jschuh, 2020-06-14:
I spent some time today just reading through the Klipper MCU code, the HX711 docs, and [this support library](https://github.com/bogde/HX711). Here's my understanding of what I need to do (this is also a note to myself, since I won't actually have time to work on this for a few weeks):

Generalize the endstop trigger code in /src/endstop.c:
- Replace the three gpio_in_read calls with a wrapper that supports a non-pin trigger.
- Similarly generalize endstop struct, command_config_endstop, etc. as needed.

Add new MCU code (e.g. /src/load_sensor_hx711.c) which implements:
- HX711 endstop threshold trigger
- host interface to set: trigger threshold, gain, and sample count
- host interface to directly read a sample (for debugging or whatever other use)

Some HX711 implementation details that stand out to me are:
- I2C protocol code is incompatible, so custom IO code is required (but it's pretty straightforward).
- Need to disable interrupts while reading individual samples, but not between samples (should be fine since blocking time will be very brief).
- Need to use something like CONFIG_CLOCK_FREQ to downscale clock for faster (32-bit) MCUs, so it's under the 20Mhz HX711 limit (but should compile-time exclude slower AVR MCUs, which don't need scaling).

@FHeilmann, does that sound about right?

Hywelmartin, 2020-06-15:
@bung69 even retract might be possible to control.. 
for a pellet extruder it would be a cost effective pressure gauge..

FHeilmann, 2020-06-15:
@jschuh sounds about right to me. Be advised that HX711 have two speeds that can be configured with a solder bridge on most boards. Moreover it may be useful to also have a "generic" interface to the HX711 for when someone does not intend to use it as an endstop (filament scales come to mind here).

Regarding the support library I don't think it's needed. If you looked through my gist, you can pretty much make the HX711 code dependency-free with very little code. Ultimately I think the low level code should be as bare-bones as possible (i.e. just get the value and forward it to the host-code, except of course during homing) and relegate as much of the 'smarts' to the python portion of klipper.

A "serious" low level change like this one  likely needs I think @KevinOConnor should give his input as well.

Bests,
Florian

jschuh, 2020-06-16:
Thanks for the feedback and the reminder about the data rate pin. And just to clarify, I didn't mean to suggest pulling in an external HX711 library. I intended that link as a note to myself, since I found it much easier to read the HX711 documentation while inspecting a full-featured implementation. That stated, Klipper won't need most of those features, and other details of the implementation will inevitably need to differ from the linked library. So, I completely agree that we would want the simplest, dependency-free implementation in the MCU code.

As I mentioned above, I'm not actually going to have time to implement and test this on a real board for another few weeks. So, I'm happy to wait for @KevinOConnor to weigh in.

KevinOConnor, 2020-06-16:
Alas, I don't know enough about the hx711 to really offer much advice.  From your comments above, I can provide some high-level tips:
1. It may be easier to introduce new mcu commands instead of trying to alter the existing endstop.c code.  (That is, enhancing the ~170 lines of endstop.c code may be more complexity than duping some of that code.)
2. I'm not sure what the communication mechanism is, but we really don't want to disable irqs for more than a few microseconds.  If the device operates something like SPI or I2C then I'd say a solution should run the comms in "task context", which will introduce greater complexity and some jitter in response time (ie, lower accuracy).

Cheers,
-Kevin

p2004r, 2020-07-05:
> The HX711 load cell ADC is used as a bed level sensor on the upcoming [Creality CR-6](https://www.creality.com/cr-6-se-3d-printer-back-us-to-save-100-p00296p1.html) series of printers. It works by [...]

The exact weight value is not required for these applications. Via a hyperlink - a simple contact detector with the same load cell. Price 4$.

https://aliexpress.ru/item/4000607444816.html?spm=a2g0o.productlist.0.0.534273f586cU2X&algo_pvid=7979999b-7802-4343-958d-d54458232393&algo_expid=7979999b-7802-4343-958d-d54458232393-22&btsid=0b8b15ea15886589429248141e4ac2&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

jschuh, 2020-07-06:
@KevinOConnor, thanks for the feedback. I had some time in the last few days to experiment with my originally proposed solution, and so far it takes too long in a busy wait loop. It looks like it could still be made to work as a bed level sensor using the task context, but the Z probing would probably have to run a bit slow to get good accuracy, so I'm doubtful of the viability of that approach as well.

It also looks like Creality's CR-6 has an MCU between the mainboard and their load cell, which handles the HX711 sampling and just emits a simple trigger (the same as @FHeilmann did with his attiny85 sample above). So, at this point it might make sense to just close this feature request out, and if necessary I can file a new one once Creality releases their patched version of the Marlin source.

That stated, I do now have MCU code for pulling HX711 samples, so if anyone is particularly interested I can see how much work it would be to port it to a task context and plumb it through to the host for other purposes (e.g. filament scales or other strain measurement). However, since my interest was in the bed level use, it's no longer at the top of my list of projects.

@p2004r, unless there's something I'm missing, I don't believe the linked module would work for the bed levelling use case. The reason for using something like an HX711 with a stock load cell is that it has multiple strain gauges in a full bridge configuration on a standard bracket. That provides consistent mounting plus reliable, repeatable measurement across a broad range of temperatures. It's also an inexpensive solution (I picked up 1kg and 5kg load cells for about $5 each US).

So, you're absolutely correct that we don't need an exact weight value for a bed level. However, we do need to be able to set a triggering threshold that is consistent across the full temperature range for a given machine, and would require minimal (or ideally no) end user configuration of the triggering threshold across machines. I'm not an expert in any of the electronics involved here, but a stock load cell seems like a good approach, and it's what Creality is doing on the CR-6 series (although their bracket is admittedly more "custom" than a typical load cell).

p2004r, 2020-07-07:
> It also looks like Creality's CR-6 has an MCU between the mainboard and their load cell, which handles the HX711 sampling and just emits a simple trigger 

The cost of HX711 is **very** low. If the installation of an intermediate controller is unavoidable, we can install 2 or 3 ADCs for one load cell. All HX711s do not work synchronously and result in a measuring current in which there are shorter (but also non-regular) gaps between the measurements. The intermediate controller coordinates the measurement flow and assigns it to time stamps.

jschuh, 2020-07-07:
Looks like I should revise my last comment. I increased the rate on my HX711 board from 10 SPS to 80 SPS, which required desoldering a resistor from one jumper pad and resoldering it onto another (as @FHeilmann had mentioned earlier). With that change I can now read a sample in under 15ms. That's still too long to have interrupts disabled, but it seems fast enough that probing speeds would be reasonable even when using an average across several samples. So, I guess I'm not giving up yet.

Before tackling the endstop implementation I plan to start with getting the basic board support in the MCU working in a task context. I'll keep that code as minimal as possible. In addition to the basic comms, the MCU will just handle things like setting the gain, taking a number of averaged samples, and resetting the HX711.

Anything more complicated (tare, calibrating with a reference weight, etc.) would be on the host side.

KevinOConnor, 2020-08-05:
I'm going to mark this as "inactive" as I don't think we need to track it as an open issue in Klipper.  If someone wishes to work on it, feel free to submit a PR when ready.

-Kevin

closed 2020-08-05 by KevinOConnor

paulhew69, 2020-10-06:
> That stated, I do now have MCU code for pulling HX711 samples, so if anyone is particularly interested I can see how much work it would be to port it to a task context and plumb it through to the [...]

I read this with interest.
I have been trying different examples of people who have written Arduino code for purely weighing filament remaining.
They have a basic database including empty spool weight, density of the filament, then it does some sums to give approx weight left and metres.
Is it possible to connect an Arduino to the Pi running Klipper, as an additional MCU, like what is done on my Voron with 2 x SKR 1.3's?

I am not a coder like your good selves but am willing to test and give input.
I do have have a friend who is competent in Arduino coding and has shown willing.

I know there are a couple of Octo-plugins that assume the HX711 is connected directly to the Pi and one that the processing is done on the Arduino and then passes the info over to the Pi. The later has 4 HX711's connected as well as a DHT sensor, fan and relay which controls a small heater for which if you are using a dry box for filament is a good solution.

I am running Klipper / MainSail and Moonraker and am so pleased I do not have to run the 'other' software.
I mentioned it to Meteyou last night and he asked if it was a Klipper function! My reply was, 'Not yet!'

So, I think the first and main question is, could a Pi running Klipper communicate with an Arduino board, connected to the USB?

Regards,

Paul.

Hywelmartin, 2020-10-07:
@paulhew69 
Yes klipper can connect arduino boards.. but only with its own firmware...

jschuh, 2020-12-04:
Sorry, I forgot to update this. I never got reliable sampling working at a high enough rate that I was happy with it for bed leveling. I have some free time coming up over the Christmas holidays, so I'll see if I can maybe clean up my HX711 code to get it checked in so people can use it for whatever purpose they like.

KevinOConnor, 2020-12-04:
FYI, there was a discussion of using load cells to probe (but not home) using an iterative measuring approach at #3496 .

-Kevin

jschuh, 2020-12-04:
Ah, thanks for the pointer. I guess if you don't need it for homing, and someone else already has code to get a reliable (albeit slow) bed leveling mesh, then it's certainly worth me cleaning up the MCU code I was working on to get non-blocking samples. I'll see if I can get that knocked out over Christmas.

mrx23dot, 2020-12-09:
Hi All, My 2 cents:

- Set speed to max 80SPS, but indicated data-ready signal is still inconsistent.
- Data can be read out super fast, at ~0.4us CLK rate  that's 10us blocking/adc reading -> not even worth more than a blocking for loop
- I would still use an intermediate MCU for modularity and provide digital out.
- Needs reset (via CLK) if data ready times out
- For best result you need differential (2) load cells, preferably at the same temp.
- Absolute value changes a lot, so I would use a slow avg and fast avg filter with cross-over comparator, and have a potmeter to set rates, also a rate of change detector in parallel.
- No need to calibrate to mass with these detectors, only to a linear scale if needed.
- calibrate at cold, with hotend off (no EMI)

CR-6:
[image]

I'm looking into adding strain gauge to an XYZ frame, but mounting is a nightmare.
That's why I first try TMC2209 sensorless, with zero extra cost. (does anyone know about this in Klipper?)

mhier, 2021-01-04:
> Sorry, I forgot to update this. I never got reliable sampling working at a high enough rate that I was happy with it for bed leveling. I have some free time coming up over the Christmas holidays, so [...]

Are you actually able to get precise enough results with a simple threshold on the load cell measurement? With my setup (different ADC, probably also different load cell and mechanical setup) this would be quite out of the question. At least if you start doing thinks like measuring a hot bed (because the shape of the bed is actually different depending on temperature!) the load cell reading tends to be drifting like hell. I would have to put the threshold so high that the result would be wrong by 0.1mm or even more - and this offset would strongly depend on temperature and random behaviour.

This is the reason why I developed the more complicated measurement code in #3496. Actually I have not invented the principle. The vendor of my 3D printer (Renkforce RF1000) has implemented this already (in a much simpler form) into the original Repetier-based firmware. A community around my printer as refined the algorithm over several years in the Repetier-based firmware, and I have used this experience for the development of the new Klipper-based code.

Maybe the optimal solution is a hybrid of the two approaches: Most time in my solution is used for finding the coarse contact position. In the long run we could maybe replace this with your MCU-based method. This could be followed by a fine scan for higher precision.