Klipper3d/klipper #2978: How do I adjust delta height after extended delta calibration

280razer, 2020-06-11:
Performed the Delta Calibration using the probe on my Kossel Linear Plus and the height was perfect.
Did the extended Calibration and now my zero point is .52 off?

How do I adjust this without messing up the calibration?

TIA

280razer, 2020-06-11:
Sorry, didnt include it because I didnt think it was necessarily an error, but here it is.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4762171/klippy.log)

BlackStump, 2020-06-11:
add SET_GCODE_OFFSET Z=? to your start gcode

best option is to set your probe Z offset so after any calibration there is no need for a z adjustment.

another option seeing you have I presume added the correct position_endstop to each stepper is to adjust those figures the same amount on each stepper to account for your Z offset

here is the tricky part do not use bed mesh until you have a correctly calibrated Delta.

Hywelmartin, 2020-06-11:
try without mesh bed leveling
The delta calibrate assume that there is a flat bed, so if you have a bowed or in any other way non flat bed it will not work good..

280razer, 2020-06-11:
When I did the initial Calibrate, Bed_Mesh had not been done nor was it active. The physical setup on the machine is as close as I can get. Like I said , it was printing very well right after the basic calibration but I noticed there was a slight warp in the build plate adjacent to the Y tower and the bed mesh was helping with that. Did the extended calibrate measurements and now the machine thinks I have an extra .52 of build volume.

I thought about just lowering the value for each endstop, but I wanted to make sure there wasn't a more preferred method first. 

I've had this Delta for about 3 years with Marlin and I hate Marlin for these things. So far, I love this Klipper Octopi method. Might even clone it on the Pi and use it for my other machine (CR-10).

Thanks again for the help

KevinOConnor, 2020-06-14:
Any time you rerun delta_calibrate, then you should also rerun the mesh calculations - as a change to the delta configuration is likely to invalidate any previous mesh measurements.

Also, be aware that almost all automatic Z probes work poorly on a delta (as explained in the documentation).  They often lead to confusing and poor results.

-Kevin

closed 2020-06-25 by KevinOConnor