Klipper3d/klipper #2979: S-Curve: float division by zero

PerLangkjaer, 2020-06-11:
Hello All

I'm trying out the S-curve branch (I know you are not keen on supporting other branches.)
But i hoped you could help me out anyways...

I get this:
Recv: // Internal error during connect: float division by zero
Recv: //
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: // Klipper state: Not ready
Recv: !! Internal error during connect: float division by zero

I can read the klippy.log but as i'm not a programmer im not the the position to correct the error :-(
Is it just a new install thats needed??

[klippy.log](https://github.com/KevinOConnor/klipper/files/4766819/klippy.log)

dushyantahuja, 2020-06-11:
hmm looks like a bug in the smooth_xy section. the documentation calls for smoother_freq_x and smoother_freq_y variables, but the code in that branch is looking for smooth_x and smooth_y. Do you want to try to add these two variables in your config section:

```
[smooth_axis]
smooth_x: ... # 2 / (3 * f_X)
smooth_y: ... # 2 / (3 * f_Y)
``` 
Also - best to report this bug on issues #57 or #2030

PerLangkjaer, 2020-06-11:
I'll give it a go tomorrow.
I needed to start a 12h print ;-)

I will report there also, thanks.

dmbutyugin, 2020-06-11:
@PerLangkjaer It should be fixed now, thanks for raising this too. Ideally yes, please report issues related to S-Curves in #57.

PerLangkjaer, 2020-06-12:
Can you please give me a simplified version of how to sync it?
Just follow the guide and pull it again?

dushyantahuja, 2020-06-12:
@dmbutyugin looks like the calculation for `accel_comp_x` is still using `smooth_x` 
```
'accel_comp_x', (3. * smooth_x / (4. * math.pi))**2,
                minval=0., maxval=MAX_ACCEL_COMPENSATION)
```

dushyantahuja, 2020-06-12:
That's probably the easiest.

dmbutyugin, 2020-06-12:
> @dmbutyugin looks like the calculation for `accel_comp_x` is still using `smooth_x`
> 
> ```
> 'accel_comp_x', (3. * smooth_x / (4. * math.pi))**2,
>                 minval=0., maxval=MAX_ACCEL_COMPENSATION)
> ```

@dushyantahuja That's a default value and is intentional. If the value is provided, default value is ignored. The problem with the other calculation was that even if the `smoother_freq_x` was provided (so default was not needed), the default was still computed, resulting in division by zero errors. Here it will not cause any troubles.

KevinOConnor, 2020-06-25:
I'm going to close this as it looks like the original issue was addressed.

-Kevin

closed 2020-06-25 by KevinOConnor