Klipper3d/klipper #2983: Klipper modifications

280razer, 2020-06-12:
Fairly new to Klipper and Octopi world. I was trying to figure out how to make a change to the system. I looked through the help stuff and on here and still cant find my answers.

When I start a print and the temps have come up the system takes a really long time to begin printing. The extruder temp goes up and down +-2 degrees from target the whole time. Is there a way to decrease the sensitivity or to reduce the time etc?

280razer, 2020-06-12:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4773118/klippy.log)
attached log

jakep82, 2020-06-12:
Have you done a PID tune?

280razer, 2020-06-12:
I did it twice at a temp of 220

Gave me almost the same exact numbers both times. The temp floats up and down a +-1 when printing, but does about +-2 while its waiting for the print to begin no matter the temp from 200 to 240

jakep82, 2020-06-12:
Trying limiting max_power to something like 0.7.  If you have a powerful heater it can overwhelm the PID loop and cause significant fluctuations.

280razer, 2020-06-12:
would I put that in the extruder section of my printer.cfg?
Also, is there a master list of all the Klipper specific commands somewhere?

Hywelmartin, 2020-06-15:
> would I put that in the extruder section of my printer.cfg?
> Also, is there a master list of all the Klipper specific commands somewhere?

Yes see here https://github.com/KevinOConnor/klipper/blob/35bbd989e4383193b8ef4cfef0b262831e611f75/config/example.cfg#L152

type help in terminal to see most commands that are available

280razer, 2020-06-25:
That's what I assumed. Any value below .8 and it never heats up. 0.9 and it has trouble keeping temp and eventually errors out.
I'll just live with the 2-3 degree swings until I upgrade to a V6 with a sock.

Thanks

KevinOConnor, 2020-07-10:
Okay, I'm going to close this then as it doesn't look like a klipper software issue.

-Kevin

closed 2020-07-10 by KevinOConnor