Klipper3d/klipper #2984: Support for DSB18B0 one-wire environmental temperature sensors

3dlpm, 2020-06-13:
Hi,

Klipper has support for environmental temperature sensors (I2C BME280 and HTU21D  two-wire sensors). Can support be added for Maxim (formerly Dallas) DS18B0 one-wire interface temp sensors?

Datasheet: [https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf](https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf))

Thank you!

3dlpm, 2020-06-13:
There is no klipper log file. This is an enhancement request

KevinOConnor, 2020-06-14:
I'm sure it's possible; an interested developer would need to implement and submit it.

-Kevin

KevinOConnor, 2020-06-25:
I'm going to close this as I don't think we need to track this as an open item.  Should someone wish to take on this work, feel free to submit a PR for it.

-Kevin

closed 2020-06-25 by KevinOConnor

SpaceEggs24, 2020-12-14:
I would also be interested in this. I would love to see something similar to the Octoprint enclosure plugin just for Klipper and Fluidd.. I would like to be able to connect multiple DS18b20 to the Pi and control some enclosure fans using a relay through the measured temperature. Maybe someone likes the idea too and has the skills to create something like that?

theopensourcerer, 2020-12-15:
@SpaceEggs24 I did try to get something to work, with a lot of help from @KevinOConnor, but unfortunately I don't really have sufficient knowledge  or proficiency in C to get any further at the moment. If you want to have a go please fell free :-)

The most recent discussions branch & MRs are here:

https://github.com/KevinOConnor/klipper/pull/3462

SpaceEggs24, 2020-12-15:
@theopensourcerer thanks for your answere! Unfortunately i probably have way less experience in C than you have so having a go at it myself is kind of out of question.. I just wanted to let developers know that there are other people interested in this topic aswell. Some kind of a plugin manager would be also nice for klipper but i dont know if thats even possible. If the functionallity would be there, it would be a no brainer to go with klipper but as of now im reluctant to make a complete switch because i'm missing out on those plugins.

theopensourcerer, 2020-12-15:
@SpaceEggs24 I think the plugin side might probably be best handled by the new "moonraker" API that was recently developed:

https://github.com/Arksine/moonraker

I've been using Klipper for over a year now, and recently moved from Octoprint to Mainsail. As far as I am concerned the ONLY thing I am missing is support for the DS18B20 on my RPi to control my enclosure fans :-)

SpaceEggs24, 2020-12-15:
@theopensourcerer I just dove into klipper (fuiddpi) yesterday and im really liking it so far. I dont know much about the software side of things but it would be awesome to have some kind of dropdown menu like in the Octoprint enclosure plugin to not have to create what you want using code. This is the plugin im referring to: https://plugins.octoprint.org/plugins/enclosure/
And yes DS18B20 support would be awesome. But what i read is that its supported by the RPi's out of the box it would just have to be integrated into the frontend? I did already use these sensors on my Octoprint install and it worked great. 
I hope there will be done some further developement on this. :)

SpaceEggs24, 2020-12-15:
> @SpaceEggs24 I think the plugin side might probably be best handled by the new "moonraker" API that was recently developed:
>
> https://github.com/Arksine/moonraker
>
> I've been using Klipper for over a [...]

Just realized that you wrote DS18B0 in the title but i guess you meant the DS18B20?

theopensourcerer, 2020-12-15:
@SpaceEggs24 Yes - I used the Enclosure plugin with the DS18B20. Yes, there is a 1-wire kernel module so not much code necessary to just "read" one of them. But AIUI Klipper is very time-sensitive and so trying to do this "within" the Klipper code itself requires some extra code to handle the fact that the sensor/kernel module is very slow to respond to each read request.

This isn't my ticket - I didn't write the title ;-)