Klipper3d/klipper #2985: Step lock

Coffee0297, 2020-06-13:
Hello, i have head that klipper don't step lock? i have noticed this bc i have a full metal printer that resonates like a mofo with klipper and not with marlin. atm i have it masked via dampers but its not really viable. 
is there any plans to make klipper step locked? 

dummy log
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4774597/klippy.zip)

Kind regards Tonny

DeviousPenguin, 2020-06-13:
Can you please explain what you mean by step lock? I tried a DDG search and got nothing, thanks :)

BlackStump, 2020-06-13:
Please supply a log from the resonating printer.
You made a statement that marlin doesn't resonate have you tried to emulate marlins parameters in klipper. 
Imho you need to compare apples to apples. Marlin to Klipper is never going to be apples to apples.
but at the very least try to have the same accel, square_corner (marlin jerk)etc etc
A printer resonating is more a product of the printer build.
I too have no idea what step lock is so please explain.

Hywelmartin, 2020-06-14:
I had some problems with resonating during quick moves, and it turned out that it was stealth chop that messed it up.. so I'm running spread cycle all the time..

Coffee0297, 2020-06-16:
so, afaik step lock is when all steppers move at the same time. i have just heard that klipper don't.
that is moves x-y-z-e and so on. only exsplanation i can find. here is a vid from marlin, 
[VID_20200603_184901.zip](https://github.com/KevinOConnor/klipper/files/4787542/VID_20200603_184901.zip)

and here is from klipper.
[VID_20200608_192750.zip](https://github.com/KevinOConnor/klipper/files/4787546/VID_20200608_192750.zip)

all i can say they ran the same acel and "jerk". stepper drivers are also same setup. i have it maskd by dampres atm. 

kind regards Tonny

jakep82, 2020-06-16:
It sounds like interpolation is turned on in the first video and turned off in the second, but you've provided no log or information about your printer so it's not possible to offer meaningful advice.

Coffee0297, 2020-06-16:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4788642/klippy.log)

interpolation is on in both instances

jakep82, 2020-06-16:
When did you last update Klipper?  A bug that prevented stealthchop_threshold from being set properly was fixed in February.  You can also run `DUMP_TMC STEPPER=stepper_x` and `DUMP_TMC STEPPER=stepper_y` to verify that both drivers are communicating properly and are running in stealthchop mode.

Coffee0297, 2020-06-16:
all that is allready checkd, i have gone trugh each register and ive also tried to set optimal parameters for my motors no change. 

im on g787ed452 (May 5)

Kind regards Tonny

KevinOConnor, 2020-06-22:
I don't understand what is being reported here.  I also looked at both videos and I don't understand what is incorrect with the second video.

-Kevin

Coffee0297, 2020-06-22:
the noise are the issue, i exspected it to sound like video 1. am i right that klipper sends step to x then y then z then e? like not 100% at the same time? 

kind regards Tonny

KevinOConnor, 2020-06-22:
Alas, you haven't attached a complete log, so there's little we can do to assist.

-Kevin

closed 2020-06-22 by Coffee0297

th33xitus, 2020-06-24:
> so, afaik step lock is when all steppers move at the same time. i have just heard that klipper don't.

> am i right that klipper sends step to x then y then z then e? like not 100% at the same time?

I can't really imagine that klipper is written in a way that multi-axis movement commands shouldn't be executed 100% synchronous. As a best case scenario of course...

Where did you heard that? Would be nice to have a source for that statement. So the devs can either confirm or disprove the rumor.