Klipper3d/klipper #2986: klipper wont connect w octoprint

wjg68, 2020-06-13:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4774673/klippy.log)  

having trouble connecting klipper to my printer. endeer 3 , silent board 1.1.5, bltouch 3.0, rasp pi 3b+ w ir cam, marlin 2.0

first post and a newbie. hope i did that right.

danis101, 2020-06-14:
Klipper is software that you upload on your printer, then you connect from octoprint to the klipper
As I understood from your post, you want to connect klipper to marlin? Can't be done

Hywelmartin, 2020-06-14:
You have to install klipper on the printerboard and configure the printer.cfg correctly  
https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md

schnello, 2020-06-17:
>Unable to open port: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'

There is no device with /dev/ttyACM0. 
Search for the correct device with ls /dev/serial/by-id/*

wjg68, 2020-06-21:
15:27:10 Printer is not ready
The klippy host software is attempting to connect. Please
retry in a few moments.
Klipper state: Not ready
15:27:10 Printer is not ready
[klippy.log](https://github.com/KevinOConnor/klipper/files/4809616/klippy.log)
still having same issue. changed 
thought i changed the printer.cfg and saved correctly but not sure. can connect to octoprint but when i check klipper status i get this error.

schnello, 2020-06-21:
Please post the output of
 ls /dev/serial/by-id/*

wjg68, 2020-06-21:
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

wjg68, 2020-06-21:
in some of the instructions the say to use the config from the raspberry pi and some say to get it from github. are they the same? I may be using one with the other by now. tried both.

wjg68, 2020-06-21:
tried sending my printer.cfg file but cant figure it out. drag and drop wont work.

schnello, 2020-06-21:
ATM the config is not your problem:
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

What we expect is:
A device with "klipper" in the name. e.g:
nas@service-uplink:~$ ls /dev/serial/by-id/
usb-Klipper_stm32f407xx_2A00390008504D4E48333920-if00

So from my postion i guess you dont have flashed the klipper.bin to your board?

schnello, 2020-06-21:
add comment:
You wrote in the config:
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0**0**

But regarding the last post the device is:
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

wjg68, 2020-06-21:
# This file contains common pin mappings for the 2018 Creality
# Ender 3. To use this config, the firmware should be compiled for the
# AVR atmega1284p.

# Note, a number of Melzi boards are shipped with a bootloader that
# requires the following command to flash the board:
#  avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex
# If the above command does not work and "make flash" does not work
# then one may need to flash a bootloader to the board - see the
# Klipper docs/Bootloaders.md file for more information.

# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: PD7
dir_pin: !PC5
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^PC2
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PC6
dir_pin: !PC7
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^PC3
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: PB3
dir_pin: PB2
enable_pin: !PA5
step_distance: .0025
endstop_pin: ^PC4
position_endstop: 0.0
position_max: 250

[extruder]
max_extrude_only_distance: 100.0
step_pin: PB1
dir_pin: !PB0
enable_pin: !PD6
step_distance: 0.010526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA7
control: pid
# tuned for stock hardware with 200 degree Celsius target
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PD4
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA6
control: pid
# tuned for stock hardware with 50 degree Celsius target
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: PB4

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[display]
lcd_type: st7920
cs_pin: PA3
sclk_pin: PA1
sid_pin: PC1
encoder_pins: ^PD2, ^PD3
click_pin: ^!PC0

wjg68, 2020-06-21:
figured it out, thats my printer.cfg

schnello, 2020-06-21:
Did you flash the klipper.bin to you printer board?

wjg68, 2020-06-21:
if it makes a difference its flashing to a 1.1.5 silent board w/ latest octoprint working ( no klipper) and an infra red camera witch isnt connected while disassembled but last it was was working. screen isnt working right now blank, kind of geryish but was after one or two of the attempts.

wjg68, 2020-06-21:
no, meant i figured how to post my printer.cfg. Newbie. 
have not flashed the klipper bin . dont really know how. is it in the directions on the klipper install?

schnello, 2020-06-21:
Without the firmware at the printer board... klipper cant connect.
Please read and study the installation instruction:
https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md

wjg68, 2020-06-21:
i think i figured out what you mean, just saw the installing DWCS after you hit the next button. Did not see that. was off my screen.

wjg68, 2020-06-21:
getting my instructions mixed up here, i think i followed all of the kevin o conner install , i found a next button that takes you to another page to install dwc on the klipper website install. havent done that yet.(dwc install)

wjg68, 2020-06-21:
i was calling myself a newbie schnello not you, thanks for help.

wjg68, 2020-06-21:
i figured it out, well **schnello** did actually. had O0 in octoprint and 0 at end of make device# it connected w klipper. see how rest goes. thank you very much everyone for all your help.

KevinOConnor, 2020-06-25:
I'm going to close this as it looks like the original issue was resolved.

-Kevin

closed 2020-06-25 by KevinOConnor

wjg68, 2020-06-26:
Thanks for your help. Having issue w bltouch setup. When I have time I'll try again and if I cant figure it out I will be back. Thanks again for your help.✌