Klipper3d/klipper #2987: Ender 3 / 3D Touch - Failed to verify BLTouch probe is raised

federov31, 2020-06-13:
Hello,

I'm using a [SKR Mini E3 1.2](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/The%20Notement%20of%20BTT%20SKR%20MINI%20E3%20V1.2.pdf) with a Creality Ender 3 Pro and a [3DTouch from Trianglelabs](https://www.aliexpress.com/item/32832887426.html)

I've been printing all day and my BLTouch was ok, when I launch my `BED_MESH_CALIBRATE` it keeps displaying :

```
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: !! BLTouch failed to raise probe
```

Can anyone help please ? 

You can find my logs and config attached
[klippy.log](https://github.com/KevinOConnor/klipper/files/4775528/klippy.log)

federov31, 2020-06-14:
Gave it another try with **v0.8.0-582-g467e8e6f-20200610_151351-octopi**

Here's the Klipper.log 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4775703/klippy.log)

madsi1m, 2020-06-14:
Hi, I use skr mini 1.2 e3 with an Ender3 and a 3D Touch. Maybe some of my settings might help: https://github.com/madsi1m/MD_Klipper/blob/master/printer.cfg

federov31, 2020-06-14:
I'll give it a try, why did you set `step_distance: 0.003125` because default for Creality is `step_distance: .0125` ?

madsi1m, 2020-06-14:
With step distance I noticed a few things, first the mcu is mostly idle during a print, so there is room for it to process more commands. Second I experimented with some test prints and it seems like having more steps gives me a cleaner print which implies that the interpolation on the tmc2209 are not as good as Klipper sending more commands to the mcu.

With the 3D Touch also double check the connector plugs on the mcu, mine were a bit loose, maybe yours is also. Other thing I suspect is the +5V line on the skr could be a little dirty on mine, but yours is a Pro with a better PSU.

How did you connect the bltouch to the skr? Is it in the z-probe pin?

federov31, 2020-06-14:
Hey @madsi1m , thanks for your time ! 

I wired as described in my first post. I also figured that the connector is a bit loose. I'll give it a try

[image]

madsi1m, 2020-06-14:
Somewhat off topic, but i also got sensor-less homing working for the x+y. (I can see you have the x+y endstops plugged in). Something to play with once the z is fixed.

federov31, 2020-06-14:
I've done the wiring again , and now the probe is working fine :). The wire leading to the Z stop was damaged ! 

Concerning the X&Y sensorless homing due to the fact that I was previously printing with Klipper between 150 and 200 mm/s, I would rather not using them because I'm afraid of skipping steps. What do you think ?

madsi1m, 2020-06-14:
Glad you found the problem.

With sensor-less homing, should not matter as the "bump" is only detected during homing. Also adjust `driver_SGTHRS` which is sensitivity. I haven't uploaded my latest yet but for my x i have it on 40 and my y is on 100. You will have to experiment so that it doesnt smash the side too hard and doesn't trigger on some dust on the wheels.

Other thing that might be of interest is the `gcode_button kill_switch` I use the original z endstop for this and have set it lower than the 3d touch so it triggers only when the 3d touch fails and the nozzle hits the bed.

federov31, 2020-06-16:
I'll get back to Klipper now that my BL is working.
Would you mind sharing your working config on Git so I can give it a try ? @madsi1m

madsi1m, 2020-06-16:
Updated, take only what you need, as things like z-offset and pid number will probably be different.

federov31, 2020-06-17:
Hello I'm using your configuration right now and it's cool. I had to change the value for the extrudeur because it was skipping steps and clicking a lot. 

Is there a way to calculate those values ? Also is it possible to calculate the vref ?

closed 2020-06-24 by federov31