Klipper3d/klipper #2989: Cannot connect to Klipper

thefabricator03, 2020-06-14:
Hi All,

I am having trouble setting up my BigTreeTech GTR V1.0 to run Klipper and DWC2.

I had a bit of playing around with the printer.cfg file and now I cannot figure out where to go from here. Still getting a error message.
MCU 'mcu' config: BUS_PINS_i2c1=PB6,PB7 BUS_PINS_i2c2=PB10,PB11 BUS_PINS_spi3a=PC11,PC12,PC10 BUS_PINS_spi2a=PC2,PC3,PB10 STEP_DELAY=2 BUS_PINS_spi3=PB4,PB5,PB3 ADC_MAX=4095 RESERVE_PINS_crystal=PH0,PH1 CLOCK_FREQ=168000000 BUS_PINS_spi1=PA6,PA7,PA5 RESERVE_PINS_USB=PA11,PA12 MCU=stm32f407xx STATS_SUMSQ_BASE=256 BUS_PINS_spi1a=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13
MCU error during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 141, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 553, in _connect
    config_params = self._send_get_config()
  File "/home/pi/klipper/klippy/mcu.py", line 541, in _send_get_config
    self._name,))
error: Can not update MCU 'mcu' config as it is shutdown

[klippy.log](https://github.com/KevinOConnor/klipper/files/4775843/klippy.log)

KevinOConnor, 2020-06-14:
This issue tracker is exclusively for use with code from this repo.  If you're having an issue when using DWC2, you'll either need to reproduce the issue on the pristine code from this repo or report it to the dwc2 developers.

-Kevin

thefabricator03, 2020-06-14:
The issue, as far as I can tell is with Klipper. Not DWC2.  I am not a programmer, but from my untrained eye it looks like the log is saying its a issue with the machine control unit, is that not correct?

Hywelmartin, 2020-06-15:
> The issue, as far as I can tell is with Klipper. Not DWC2. I am not a programmer, but from my untrained eye it looks like the log is saying its a issue with the machine control unit, is that not correct?

In that case it will be a easy to reproduce the issue on pristine code from this repo....

thefabricator03, 2020-06-15:
Lol @ pristine code - do you really have your head that far up your ass?

thefabricator03, 2020-06-15:
This is the code I used to install Klipper,

git clone https://github.com/KevinOConnor/klipper
./klipper/scripts/install-octopi.sh

Does that mean I am using your "pristine code" ?

thefabricator03, 2020-06-15:
I have herd a lot of good things about Klipper, but the arrogance I am getting from the Author makes me think otherwise. Can someone please explain the damm error message I am getting in the log? Why is that so hard?

BlackStump, 2020-06-16:
[Official Site](https://www.klipper3d.org/) 
 
Self help is a wonderful thing, your attitude is not going to get you far.

thefabricator03, 2020-06-16:
I have searched for what that error means, spent 6 hours on it actually. Frankly how hard is it to answer a simple question. I have loaded my log file.I have spent time going through the doc in the Klipper folder, looked in every part of the internet for similar cases and I still cannot get a answer. 

Is this one that just gets put in the too hard basket? Or can is there a way to find out what is happening? I am all ears and I am willing to try anything. I am not asking for the Author to come to my facility and fix it himself, all I need it a bit of guidance to point me in the right direction.

thefabricator03, 2020-06-16:
I have used the logextract.py script to generate these new logs. Does anyone have any idea on why the MCU is shutting down?

[klippy.log](https://github.com/KevinOConnor/klipper/files/4783869/klippy.log)
[klippy.log.config0001.cfg.txt](https://github.com/KevinOConnor/klipper/files/4783871/klippy.log.config0001.cfg.txt)

BlackStump, 2020-06-16:
I have no idea what your problem is I do not have that board, maybe not to many people do at a guess.
I have a skr pro and that works fine, same chipset as far as I know.
I can only offer a the sledgehammer approach
do a clean install of your os
follow the official instructions to the letter
when you make menuconfig make sure you only change what is required.
make sure the sdcard is in good condition to upload the firmware
again follow the instructions to the letter nothing less and definitely nothing more.
maybe your board is faulty -  have you considered that?

thefabricator03, 2020-06-16:
Thank you, 

I will give it a go,

I have considered the possibility of the board being faulty. I have a new SKR pro I could try if I cannot get it to work after a clean install.

I appreciate the advice.

jakep82, 2020-06-16:
Rather than continuing to spend hours trying to make DWC2 work, you could simply install Klipper on Octopi using the official documentation and submit a log with pristine code as requested.  Without pristine code it's impossible for the developers to know whether the issue is with Klipper, your hardware, or the code you've modified to install DWC2.

thefabricator03, 2020-06-16:
That might be the way to go. I will try Octopi and see how I get on.

I really did not think it would turn out like this. I have a bit of experience with LinuxCNC and for a open source project it generally works pretty well. I guess I am just having a run of bad luck, but at least I am learning the hard way.

thefabricator03, 2020-06-18:
Installed Octoprint and now its woking. Sorry for getting pissed off, it can be frustration when things dont work like you think they should. Klipper is working perfectly - its very similar to my LinuxCNC CNC Plasma setup. Thanks Everyone who commented.

closed 2020-06-18 by thefabricator03

Hywelmartin, 2020-06-18:
Be a little more patient next time.. and more people will be at your service...
Glad that it worked out...

Shine6Z, 2020-07-22:
Recv: // MCU 'mcu' shutdown: ADC out of range
Recv: // This generally occurs when a heater temperature exceeds
Recv: // its configured min_temp or max_temp.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'mcu' shutdown: ADC out of range