Klipper3d/klipper #2990: poor layers quality when using jerk or acceleration

danis101, 2020-06-14:
Hi Im fighting with poor walls quality when I use cura with jerk and acceleration control
turning only jerk results as well with poor walls, also turning only acceleration control results with bad layer quality.
Not using those options provide ghosting (after corners waves) but layers are perfect - here is photo what I mean: https://www.dropbox.com/s/3jx31wlfpg25auj/104232971_930928014091501_962199983343950252_n.jpg?dl=0
Left one is without jerk and acceleration control (turned off)
Right print is with only jerk (only acceleration results the same)
configurations: https://pastebin.com/yD7M4rif
klippy.log : https://pastebin.com/9UdSmTKB
Can someone tell me what is going on with this?

danis101, 2020-06-14:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4776301/klippy.log)
Don'touhh actually I pasted klipper log in pastebin

Hywelmartin, 2020-06-14:
jerk_control does noting

danis101, 2020-06-14:
so how I am supposed to get rid of ringing?
Belts are tight, verry tight

Hywelmartin, 2020-06-14:
Set acceleration: M204 S<value> OR M204 P<value> T<value>

    Note: If S is not specified and both P and T are specified, then the acceleration is set to the minimum of P and T. If only one of P or T is specified, the command has no effect.

For Jerk.. or the similar ... SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=<value>
so a macro is needed if you is interested to use the slicer value

danis101, 2020-06-14:
> Set acceleration: M204 Sxxx OR M204 Pxxx Txxx
> 
> ```
> Note: If S is not specified and both P and T are specified, then the acceleration is set to the minimum of P and T. If only one of P or T is specified, the command has no effect.
> ```
> 
> For Jerk.. or the similar ... SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=xxx
> so a macro is needed if you is interested to use the slicer value

Thanks, Im gonna test it out
So using slicers jerk and accelerations is  pointless?

danis101, 2020-06-15:
Okay Changed in config those settings Also deleted M20X from CURA start GCODE 
Looking verry promising, I decreased also Current some current in case of some random lines in high walls, now they are looking good, what Ive noticed if I open the doors from room where printer is located Lines sometimes appear -  looks like PLA dont like rapid temperature changes.

Hywelmartin, 2020-06-15:
I use Prusaslicer and g-code flavor Rep Rap/Sprinter to get acc control of each type of movement
[image]

For jerk I simply have a string in the start g-code
[image]

danis101, 2020-06-15:
I did all that and there is some random lines like in picture: https://www.dropbox.com/s/t7uqfk0c616yg5c/103972263_550951625788128_6630703793569336800_n.jpg?dl=0
May it be caused by overheating of TMC / Steppers? They are in SKR in original box from cr10 with 2 fans, motors are about 50-60 Celcius

KevinOConnor, 2020-06-25:
FYI, it sounds like your questions are more general 3d printing questions, and not really specific to Klipper.  You may find a wider audience for your questions in a 3d printing forum.  Your last question is commonly referred to as "z banding" and there are a number of causes for it (typically hardware related).

-Kevin

danis101, 2020-06-26:
> FYI, it sounds like your questions are more general 3d printing questions, and not really specific to Klipper. You may find a wider audience for your questions in a 3d printing forum. Your last [...]

This is not about Z banding - threads are straight and clean, inconsistances are not looking like "layer shifts" but its like Wider extrusion or smth likee that, it starts to show above 20mm/s and up outer wall speed, it is not showing when printer uses big accelerations and cornering speeds, If I put theem a little down to get off with ringing/ghosting Im getting this result, so printing outer walls under 20mm/s seems like solution for now. With klipper it is a little better than marlin even with SKR 1.3 board so those could be:
-too much steps on layers
-too fast Z acceleration or speed causing loosing steps that gives thinner layer than it should be, so extrusion is wider
- or just a bug with lower accelerations (700 or less)

anyways for now I dont think it is a klipper problem and you might be right it can be mechanical problem but definetly not Z rails or threads

changedsoul, 2020-07-04:
As a piece of info I have found out, it may be your extruder skipping steps. I had my retraction speeds to fast and it skipped on retracts and returns. Sometimes is would miss on return causing under extrusion, sometimes it would skip on retract and not on return and would cause over extrusion.  Slowing down retracts helped a lot. 
Not sure if this is what your dealing with, but its worth a look.

closed 2020-07-25 by KevinOConnor