Klipper3d/klipper #2993: safe_z_home not working with bltouch

therealjagliano, 2020-06-15:
Hi,

I am having issues with safe_z_home and using the bltouch v3 as a probe. When I attempt to safe_z_home, the printer successfully performs the x home, y home, and initial z probe. After the initial z probe, the printer attempts to redeploy the probe without moving the z axis. As a result, the bltouch probe cannot redeploy and fails homing.

Thanks!

[klippy.log](https://github.com/KevinOConnor/klipper/files/4782882/klippy.log)

therealjagliano, 2020-06-15:
Here's some snippets from my config file:

[safe_z_home]
home_xy_position: 25,25
speed: 40
z_hop: 20
z_hop_speed: 5

[bltouch]
control_pin: P2.0
sensor_pin: ^P0.10
z_offset: 2.0

[stepper_z]
step_pin: P0.22
dir_pin: !P2.11
enable_pin: !P0.21
step_distance: 0.000390625
endstop_pin: probe:z_virtual_endstop
position_endstop: 0.0
position_max: 180
homing_speed: 5
second_homing_speed: 1
position_min: -2

BlackStump, 2020-06-16:
under stepper_z
take out your
[position_endstop: 0.0](https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L558)
and
second_homing_speed:1 
instead 
[use](https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L654)
options under bltouch ie
>speed: 1.0
>samples: 3
>sample_retract_dist: 5

therealjagliano, 2020-06-16:
So strangely after a power cycle and attempting to home again my config started working. I changed the options to what you have suggested and everything works. thanks!

therealjagliano, 2020-06-17:
ok this is bizarre. had this working all day today. I even was able to make a few prints with the above recommended config changes. I power cycled everything and now I am getting the same behavior as before. I confirmed that I have the config set up as recommended.

[printer.txt](https://github.com/KevinOConnor/klipper/files/4790084/printer.txt)

BlackStump, 2020-06-17:
sounds more like a wiring problem, intermittent bltouch power problem

therealjagliano, 2020-06-17:
I don’t think that the BLTouch is the issue. It triggers once, but the z axis does not retract down after the BLTouch triggers. As a result, the BLTouch attempts to deploy a second time but is blocked by the build plate. I am able to run homing with a normal optical endstop and then perform bed mesh leveling just fine with the BLTouch.

therealjagliano, 2020-06-17:
Never mind I think I understand what you’re saying. I’ll investigate further tomorrow.

therealjagliano, 2020-06-17:
I was able to get it working again by adding probe_with_touch_mode: True to the config, issuing a BLTOUCH_DEBUG COMMAND=reset, and performing a host reset. Not sure which part actually fixed the issue.

closed 2020-06-17 by therealjagliano