Klipper3d/klipper #2996: Duet Maestro Z_PROBE_IN / Z_PROBE_MCU

asolino, 2020-06-17:
I'm trying to configure an NPN Inductive Sensor and, according to the [documentation](https://duet3d.dozuki.com/Wiki/Connecting_a_Z_probe) I should be connecting the probe's signal to `Z_PROBE_IN`, and then configure klipper's `[probe]` section. However, I cannot access the `Z_PROBE_MCU` PIN (in theory `AD11`), klipper would not recognize such PIN.  According to the schematic `Z_PROBE_MCU` and `Z_PROBE_IN` are connected together.

Any help would be appreciated. Kudos for this project, love klipper.

asolino, 2020-06-17:
Attaching log...
[klippy.log](https://github.com/KevinOConnor/klipper/files/4793631/klippy.log)

KevinOConnor, 2020-06-17:
Unfortunately, the upstream Duet documentation is terrible with pin naming.  I updated the Klipper generic-duet2-maestro.cfg with some hints for the probe pins (commit 75506417).

In short, the real name of the Z_PROBE_IN pin is PC15.

-Kevin

asolino, 2020-06-18:
Thanks for your quick answer @KevinOConnor, works like a charm.

May I ask where did you get that PIN info from? I was looking at the schematic [here](https://github.com/T3P3/Duet/blob/master/Duet2/Duet2Maestro_v1.0/Duet2Maestro_Schematic_v1.0.pdf)

KevinOConnor, 2020-06-18:
The duet schematics show the micro-controller pin numbers.  The Atmel sam4s datasheet has a translation from those pin numbers to the gpio names.

If the "CR-20 LCD" is the same as the "128x64 Full Graphic Creality CR10 / ENDER 3 stockdisplay" listed in config/sample-lcd.cfg then it isn't directly compatible with the Maestro.  However, you may be able to build a custom wiring harness to connect the pins you need.

-Kevin

asolino, 2020-06-19:
> The duet schematics show the micro-controller pin numbers. The Atmel sam4s datasheet has a translation from those pin numbers to the gpio names.

Thanks a lot for the info @KevinOConnor.

> If the "CR-20 LCD" is the same as the "128x64 Full Graphic Creality CR10 / ENDER 3 stockdisplay" listed in config/sample-lcd.cfg then it isn't directly compatible with the Maestro. However, you may be able to build a custom wiring harness to connect the pins you need.

Based on some research I did, I don't think it is exactly the same but yeah, I think I'll need to do some wiring myself. 

Thanks again. Closing this issue since original problem is now working.

closed 2020-06-19 by asolino