Klipper3d/klipper #2997: SKR mini E3 DIP - diag1 pin

SrGeek, 2020-06-17:
What is pin number for the diag1 pin on an SKR mini E3 Dip to use sensorless homing.  I cant find it on any schematics.

jakep82, 2020-06-17:
The diag1 pin should be connected to the endstop pin for each axis.  You can confirm that with a multimeter.

SrGeek, 2020-06-17:
@jakep82 I checked for continuity between the pin outs for the 2130 and the endstop and there wasnt any.  I know they are normally used in UART mode.  Would the same apply in SPI mode?

jakep82, 2020-06-18:
I don't have one of these boards, but I would expect the pin connector on the SKR board for diag1 to be connected to the corresponding endstop pin.  Diag1 is it's own specific pin, and is unrelated to SPI.  The 2130 does not communicate with UART.  See [this drawing](https://learn.watterott.com/silentstepstick/pinconfig/tmc2130/tmc2130_v1.png) for the location of diag1 pin on the driver.

KevinOConnor, 2020-06-22:
FYI, if you don't see a diag pin on the schematics that generally indicates that the board manufacturer did not add the wire routing for that pin.

-Kevin

closed 2020-06-24 by klipper-gitissuebot