Klipper3d/klipper #2999: TMC driver 256 microstepping?

DroneMang, 2020-06-18:
Hi,

I have an SKR PRO.  I have tried up to 256 microstepping on Z and E, and on E there is actually a sound difference, the extruder sounds more like a squeak than a radiation detector, all with interpolate on.  I have tried up to 128 on X and Y but at higher speeds there can be skipped steps so 64 is better.  Usually, most people use these drivers with 16 microstepping and interpolate on.  I have tried the higher microsteppings in Marlin, and the print quality doesn't improve, it gets worse, brims print poorly with lots of stops and starts.

Can someone perhaps comment on if klipper will handle the higher microsteppings well?  Is there any point in trying this?  Or should I just stick with 16 microstepping?

I am just switching over to Klipper from Marlin to see if Klipper can print better with 32-bit and TMC drivers so I will test this as well.

Thank you,
DroneMang

BlackStump, 2020-06-18:
A quick google would have answered you questions ie
[1](https://www.machinedesign.com/archive/article/21812154/microstepping-myths)
[2](https://www.linearmotiontips.com/microstepping-basics/)
My understanding and it is limited is that there is very little to gain and some to lose over 16 microsteps with interpolation.
Positional accuracy is better obtain by smaller step angle stepper motors ie 1.8 to 0.9 degree steppers
noise can be improved at low speed with higher microsteps will it be better then 16 microsteps with 256 interpolation, (the key words are low speed) maybe but then again that is where stealthchop also shines at low speed. 
high microsteps also tax the mcu higher.
The limit on microsteps is not klipper it is your mcu, and that is a hardware limit
[Step Benchmarks](https://www.klipper3d.org/Features.html)

Sineos, 2020-06-18:
Actually, I was thinking about the same question also. 
In other firmwares, this is not really so dominant, because the clear limitation is the MCU. In Klipper thou this is different.

So when Klipper does its internal numerical magic it takes the mm/step value, which is depending on the mechanical characteristics as well as the microstep setting, correct?

Questions:

- Given a constant mechanical characteristic, I could "improve" Klippers numerical resolution by choosing a higher microstepping. Is this worthwhile?
- What is the difference between the TMC's "true" microstepping resolution and "interpolated" microstepping resolution, e.g. in terms of torque, precision etc?
- If interpolated to 256 microsteps anyway, wouldn't it be beneficial to have a higher "true" microstepping part and a lower interpolated one?

BlackStump, 2020-06-18:
Just a thought but would these questions be better put to TMC?
A stepper driver is just that, they drive a stepper motor.
afaik Klipper can not make a stepper motor more or less precise then what it's mechanical/electrical characteristic's are.
Choosing the best stepper for you intended purpose is important and duet3d has a good blurp about choosing a stepper for a 3dprinter. 
[here](https://duet3d.dozuki.com/Wiki/Choosing_and_connecting_stepper_motors)

It is so easy to forget the stepper motor and focus all ones attention on the driver.

DroneMang, 2020-06-18:
There did seem to be an increase in precision on the extruder. I was using
a direct drive extruder and putting my finger on the drive at 16
microstepping I could literally feel the extruder jogging or jumping to the
next step and then just sitting and doing nothing sometimes like for very
slow printing moves and infill. Printing the same part with higher micro
stepping I could sense much finer movements. For direct drive on the SKR
pro I had 256 micro stepping on e and that was printing fairly well, once I
went to a geared extruder I was getting skipped steps and ended up at 64
micro stepping.

Another issue with trying this is the slicer behavior. Simplify 3D brims
just printed horribly, the part would turn out okay though. With Marlin my
guess is there could be some limitations in terms of sending g codes to the
controller, because of the higher micro stepping the firmware can interpret
a short move in a much different way as in creating stop and start moves
where with lower micro stepping there wouldn't be enough time or space
within a movement to do that so it turns into more of a smooth move.

Of course, klipper is not Marlin and I'm thinking klipper will behave
differently in this respect.

Thanks!
DroneMang

Spacemarine2018, 2020-06-21:
I experimented a lot with klipper and different microstep-settings to get the most out of my motors. I'm not 3D-printing, but building a different kind of robot for which klipper happens to be perfect.

> So when Klipper does its internal numerical magic it takes the mm/step value, which is depending on the mechanical characteristics as well as the microstep setting, correct?

Correct.

>     * Given a constant mechanical characteristic, I could "improve" Klippers numerical resolution by choosing a higher microstepping. Is this worthwhile?

I always choose the microstep resolution to be as high as possible. There is absolutely no drawback in going as high a possible in my experience, other than high MCU load. But even that is no problem, as klipper will tell you when the MCU becomes too slow.

>     * What is the difference between the TMC's "true" microstepping resolution and "interpolated" microstepping resolution, e.g. in terms of torque, precision etc?

There is no difference in torque, as torque doesn't depend on microstep resolution, only on the angle of the rotor. Interpolation only increases the smoothness of a rotating motor, it does not increase precision

>     * If interpolated to 256 microsteps anyway, wouldn't it be beneficial to have a higher "true" microstepping part and a lower interpolated one?

Of course. The general rule of thumb is: Turn up the microsteppings as far as possible without overloading the MCU and interpolate rest.

>  I have tried up to 128 on X and Y but at higher speeds there can be skipped steps so 64 is better.

What EXACTLY do you mean with skipped steps?

DroneMang, 2020-06-21:
Skipped steps in this context was not just missing a few steps. What would
happen is the X, Y, or extruder motor would buzz / grind on the order of
seconds when commanded to move too fast. When I upgraded to a B2D geared
extruder I could no longer keep the 256 micro stepping. The extruder would
do fine for slow moves but for fast changes it would buzz/grind. Thanks!

DroneMang

Spacemarine2018, 2020-06-21:
So you had a case where the ONLY thing you changed was the microstep resolution and the result was wether you got buzzing or correct movement? That is really strange, I can't think of any reason why that should be the case.

DroneMang, 2020-06-21:
What I did to test this is to set a given microstepping, say 64, 128 or
256. (Can't get grinding at 16 or 32). Then run G0 F3000 x200 back and
forth on each axis and increase the speed higher and higher until you get
grinding, and then that is your speed limit. This was tested in Marlin, I
need to test in Klipper still. I need 200-300 mm/s travel speeds so that's
my constraint.

Spacemarine2018, 2020-06-21:
Ok, marlin might be different, Klipper should not behave like this. I suspect that the steprate was too high for marlin and steps were missed INSIDE the MCU. In that case, Klipper would exit with an error. So it has nothing to do with stepper torque.

Sineos, 2020-06-21:
> A quick google would have answered your questions ie
> [1](https://www.machinedesign.com/archive/article/21812154/microstepping-myths)
> [2](https://www.linearmotiontips.com/microstepping-basics/)

Regarding the torque: The links posted by @BlackStump indicate a quite drastic relationship between torque and microstepping.
My understanding of the worst case is: The stepper needs multiple microsteps until enough torque is "built up", e.g. to overcome mass inertia, friction etc. Don't know if you would call this "lost steps" but surely the movement will not be as expected.

Spacemarine2018, 2020-06-21:
> Regarding the torque: The links posted by @BlackStump indicate a quite drastic relationship between torque and microstepping.

No they don't, at least not in the context of this discussion. They say that the INCREMENTAL torque decreases as the step angle decreases (or microstepping resolution increases), but the absolute torque of the motor as a function of rotor orientation does not change. And that is the only relevant figure when it comes to determining wether a stepper will stall during rotation.

If you have a different optinion, please quote the exact sentence from the source, which you believe to make a different statement.

And btw.: I attended the Trinamic Webinar a few days ago, one of the participants asked exactly this question: Does increasing the microstep resolution decrease torque? The answer from trinamic was: No, of course not. Always run the highest microstep resolution possible.

Spacemarine2018, 2020-06-21:
> My understanding of the worst case is: The stepper needs multiple microsteps until enough torque is "built up", e.g. to overcome mass inertia, friction etc.

That is true! If a stepper needs 1/16 of a step to overcome the needed torque, it would need 2 times 1/32 steps, 4 times 1/64 steps, 8 times 1/128 steps or 16 times 1/256 steps to overcome this. So it makes no difference.

DroneMang, 2020-06-21:
I also wanted to point out that this discussion relates to drivers like
DRV8825 where you have to change the micro stepping with a pin, and with
those drivers changing the micro stepping does actually change the torque
behavior of the motor. I was told by my printer manufacturer to not run 32
micro stepping on X and Y because there would not be enough torque with
that driver (large format so heavy gantry).

But with TMC drivers this is all happening digitally, and with logic so I
think the torque would be maintained as you stated above.

Sineos, 2020-06-21:
I'm by no means an expert on steppers or their drivers. 
What I understand from the discussion is:

- Total torque of a stepper does not change because it is related to the mechanical and electrical properties
- The torque of a single microstep is a fraction of the above total torque, depending on the microstep resolution 

For 3d printing IMHO we rely on a simple fact: If we command the stepper to move 3 microsteps, we expect it to actually move 3 microsteps as physical motion.
Now if the mechanical load, friction etc exceeds the torque of `3 steps x incremental microstep-torque`, then simply nothing moves. This in turn would heavily affect the accuracy we are actually aiming for.

So strictly speaking, of course torque is not reduced, but we do not get the result we are expecting. Also this would explain @DroneMang's observation:

> The extruder would do fine for slow moves but for fast changes it would buzz/grind

Within the given microsteps the incremental torque is simply not enough to get the system moving at the desired speed. The stepper "fights" to do so but actually nothing moves but only buzzes.

Spacemarine2018, 2020-06-21:
> For 3d printing IMHO we rely on a simple fact: If we command the stepper to move 3 microsteps, we expect it to actually move 3 microsteps as physical motion.

I think I understand, where the problem is. You are right, if you command the stepper to move 3 (micro)steps, of course the physical result will be greatly influenced by the microstepping resolution.

But that is not a real world example. In 3D-printing you command the stepper to move the carriage by 3 millimeters. And this will result in 100 full steps or 200 half-steps or 400 quarter-steps etc.... depending on your microstepping setting. And in this scenario, it makes no difference which setting you choose, because the requested rotor angle at the beginning and end of the commanded move is exactly the same. (except for rounding errors of course)

DroneMang, 2020-06-21:
Not to complicate this further, but for the DRV8825s they sell TL smoothers
which are only helpful when at 32 micro stepping. These are for eliminating
salmon skin and such. In discussions about using those on another forum we
figured out that you may miss micro steps but not full steps. So you may
not have enough torque or precision to hit every single micro step but once
the move proceeds to the next full step you would have enough torque there.
So you would not have any artifacts in your print such as a layer shift,
just a quality issue.

Spacemarine2018, 2020-06-21:
> So you would not have any artifacts in your print such as a layer shift, just a quality issue.

Yes. And the following applies for at least all TMC drivers: If you increase your microstepping resolution, your quality will never get worse. But you might not the the expected improvement you hope for.

DroneMang, 2020-06-23:
I'll report my findings.  With 256 and 128, klipper goes into shutdown on fast moves. 256 would home at 50mm/sec, but much faster than that caused shutdown. 128 was ok up to 100 mm/s, shutdown at 200. 64 microstepping can go up to 360 mm/sec, shutdown at 380. 

I have 0.9 deg steppers on X and Y, and with 64 microstepping steps/mm is 640, and mm/step or step_distance is 0.0015625

Z can handle 256 microstepping just fine.  Nothing in the log to show the MCU was overloaded IMHO.

Thanks!
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4820107/klippy.zip)

KevinOConnor, 2020-07-10:
Any further updates on this?  It seems like the answers were found for the original questions.

-Kevin

DroneMang, 2020-07-10:
I did some testing and figured out that the best setting for me is 64 micro
stepping on X Y and E, 256 on Z. I can have travel speed up to 360 mm/s.
Higher microsteppings lead to MCU shutdown at too low of travel speeds. I
do get some strange resonances printing small circles or brims that I did
verify were not caused by the micro stepping setting, I changed back to 16
microstepping and still heard and felt the same resonance. You can probably
close this though.

Spacemarine2018, 2020-07-10:
So the findings can be summed up as: Unless you get an MCU shutdown, there is no drawback if you go to higher microstepping settings.

Maybe this can somehow find it's way into the documentation, as there always seems a lot of confusion about this.

DroneMang, 2020-07-10:
One thing I have not tested is if you will get an MCU shutdown if all axes
are moving. For example if I move X at 360 mm / s I might be fine but 370 I
will get a shutdown. What happens if I move X, Y, Z at their max speed? I
may try to test this sometime.

DroneMang, 2020-07-10:
I would like to know what the limiting factor is with this. Is it the
controller boards clock speed or is it the clock speed on the TMC drivers?
For example with the 5160s I am not sure which clock is being used or if
the drivers are syncing with the board. Is it some kind of bus speed?

Separate topic but I have considered gearing the steppers down on X and Y
for even more precise movements, in this case I would have to go down on
micro stepping to hold my travel speeds roughly the same. 0.9° steppers
have made a big difference though compared to 1.8.

KevinOConnor, 2020-07-10:
> I would like to know what the limiting factor is with this.

On an stm32f4 micro-controller, the primary limit is going to be the 2us delay between step pulses.  (For other micro-controllers, the limit is often the micro-controller processor speed - in which case one would look at the benchmarks: https://www.klipper3d.org/Features.html#step-benchmarks ).

If you're only using trinamic and/or allegro stepper drivers then you can modify the step delay in "make menuconfig" from 2us to 1us.  This should roughly double the amount of steps you can send to one single stepper.

If one is exclusively using trinamic stepper drivers and if one is using a micro-controller slower than an stm32f407 chip then it's is often possible to go to 0us in "make menuconfig" for a further increase.

-Kevin

DroneMang, 2020-07-11:
I have an SKR pro and TMC 5160s, I think that is the microcontroller you
were referencing, but when I go to make the firmware to put on the SD card
for the SKR pro there basically are no other options that I can choose, or
are you talking about something that goes on the raspberry pi? Thanks!

KevinOConnor, 2020-07-11:
To set the step pulse duration, run "make menuconfig", select "Enable extra low-level configuration options", select "Specify a custom step pulse duration", and then alter the "Step pulse duration" field.

-Kevin

Spacemarine2018, 2020-07-11:
I just looked up the specification the the TMC5160, which I am using myself on an SKR Pro Board.

According to the specification of Watterott, the CLK-Input of the TMC5160 chip is connected to ground, which activates the internal clock. Source: https://github.com/watterott/SilentStepStick/blob/master/hardware/SilentStepStick-TMC5160_v15.pdf

According to the datasheet of the TMC5160 chip,
https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC5160A_Datasheet_Rev1.14.pdf
the internal clock runs at 12 Mhz (page 122)
The maximum step rate is described at page 94, it is clock/2 = 6 Mhz if you run at 256 microstep resolution. If you decrease microstep resolution, the max step rate also goes down linearly, until you reach clock/512 = 23.4 KHz for full steps (which probably nobody runs)

Page 94 also states, that the step input low times need to be at least 100 ns, which is 1/10 of what Kevin recommended for the stm32f407.

I am running the stm32f407 with TMC5160 at 0 µs step pulse duration, and I have encountered no problems so far, at least none that I am aware of. Are there any confirmed reports of people getting into trouble when using 0µs on a fast micro controller and howdoes this trouble look like?

@KevinOConnor 
Is it possible to add a 100ns step-pulse duration setting? I feel 0 µs and 1 µs is too coarse if trinamic recommends 100 ns.

KevinOConnor, 2020-07-13:
> I am running the stm32f407 with TMC5160 at 0 µs step pulse duration, and I have encountered no problems so far, at least none that I am aware of. 

Last I checked, setting 0us results in about 120ns pulse lengths on the stm32f407, however there were reports of failures - see: https://github.com/KevinOConnor/klipper/issues/2011

> Is it possible to add a 100ns step-pulse duration setting? I feel 0 µs and 1 µs is too coarse if trinamic recommends 100 ns.

FWIW, sending more than ~100K step pulses to a single driver doesn't interest me, personally.  It's fine if someone else is interested, but that person will need to do the work and testing for it.

-Kevin

Spacemarine2018, 2020-07-14:
I can confirm, the pulse length of the step signal it is around 115 ns. Since this is within the Trinamic spec, it should work with the 0 µs setting.
[image]

KevinOConnor, 2020-08-05:
I'm going to close this as it looks like the original questions were answered.

-Kevin

closed 2020-08-05 by KevinOConnor