Klipper3d/klipper #3004: Serial connection closed

Diana6969, 2020-06-20:
I installed klipper on my stock Ender 3 today, following every step on the website. However I got into a problem, when configuring Klipper. I type restart in the terminal and it says: "Send: restart
Recv: ok
Recv: // Klipper state: Disconnect"
And then nothing :(
So then I reconnect:
10:41:28 Connected to host via /tmp/printer @115200bps
10:41:28 Firmware version: v0.8.0-605-g75506417
10:41:36 Printer is not ready
The klippy host software is attempting to connect. Please
retry in a few moments.
Klipper state: Not ready
10:41:36 Printer is not ready
10:44:01 Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
10:44:01 Unable to connect..."
PS: using octopi and used the right serial port (/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0)
[klippy.log](https://github.com/KevinOConnor/klipper/files/4807704/klippy.log)

KevinOConnor, 2020-06-22:
These types of errors generally occur when the board is not correctly flashed.  Did "make flash" report success for you?  Also, see the important notes at the top of https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender3-2018.cfg

-Kevin

Diana6969, 2020-06-24:
> These types of errors generally occur when the board is not correctly flashed. Did "make flash" report success for you? Also, see the important notes at the top of https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender3-2018.cfg
> 
> -Kevin

Well, it worked at first and I got into the problem depicted above... Then I reinstalled Klipper and yeah I can't even flash with success...
I can't flash with success following the Klipper docs/Bootloaders.md and the "avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex" command, so **I guess I'll need an arduino to flash a bootloader onto the board?**

Diana6969, 2020-06-24:
Ok, I installed bootloader and flashed Marlin with an arduino. Issue solved

closed 2020-06-24 by Diana6969