Klipper3d/klipper #3005: SKR MINI E3 v1.2 Configuration using Pins reserved for SPI?

bskp, 2020-06-20:
I was swapping my stock Creality-CR-20 Pro Mainboard with a BTT SKR MINI E3 v1.2.

After upgrading klipper to the latest version, rebuilding the firmware and copying it to the SKRs SD-card, **I'm now failing to configure klipper** using the [given template](https://github.com/KevinOConnor/klipper/blob/master/config/generic-bigtreetech-skr-mini-e3-v1.2.cfg).

Apparently, the Pins **PB14, PB15 and PB13** (used to control the X stepper) are reserved for SPI: 

`Recv: // pin PB15 is reserved for spi2`

To my understanding, I don't even need the SPI Bus 2, so I'm tempted to [remove the section reserving these pins](https://github.com/KevinOConnor/klipper/blob/755064177bd62678c9f1b731ccb3c9b29b41ce8c/src/stm32/spi.c#L19) and rebuild my firmware. However, I'm quite sure I am getting something wrong here.

Thank you for any pointers or explanations on reserved pins.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4807726/klippy.log)

KevinOConnor, 2020-06-22:
The uc1701 uses the spi bus, which defaults to spi2 on the stm32f103 mcu.  I'm not sure the display you are trying to configure will work on the mini-e3, so you probably want to remove the `[display]` section until you get everything else running.

-Kevin

bskp, 2020-06-22:
Hi Kevin,
This makes perfect sense 🤦, thank you so much for your quick help.
I‘ll try this as soon as possible and post an update!

KevinOConnor, 2020-07-10:
Okay, I'm going to close this as it looks like the original question was answered.

-Kevin

closed 2020-07-10 by KevinOConnor