Klipper3d/klipper #3007: Prints with bed mesh applied much worse than without it.

rlesac, 2020-06-20:
[2020-06-20.printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/4808042/2020-06-20.printer.cfg.txt)
[image]
LEFT with bed mesh applied, RIGHT no bed mesh
[image]

I was out of my wits trying to get a proper bed mesh, until I decide to turn it off and tried printing, lo and behold, MUCH BETTER print. 
I have no explanation for why this would happen. 
I tried calibrating the bed mesh countless times, while the mesh itself seems accurate (it changes when I adjust the bed screws) and is precise (is repeatable), Klipper seems to apply the mesh in a completely wrong way.

rlesac, 2020-06-20:
[klipper-with-bed-mesh.log](https://github.com/KevinOConnor/klipper/files/4808122/klipper-with-bed-mesh.log)

rlesac, 2020-06-20:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4808132/klippy.log)

DeviousPenguin, 2020-06-21:
What probe are you using?

rlesac, 2020-06-22:
> What probe are you using?

Genuine BLTouch.
Also, here's the probe accuracy test:
```
PROBE_ACCURACY at X:100.000 Y:100.000 Z:4.378 (samples=10 retract=2.000 speed=0.5 lift_speed=0.5)

probe at 100.000,100.000 is z=2.380000
probe at 100.000,100.000 is z=2.385000
probe at 100.000,100.000 is z=2.382500
probe at 100.000,100.000 is z=2.380000
probe at 100.000,100.000 is z=2.380000
probe at 100.000,100.000 is z=2.382500
probe at 100.000,100.000 is z=2.382500
probe at 100.000,100.000 is z=2.385000
probe at 100.000,100.000 is z=2.385000
probe at 100.000,100.000 is z=2.385000
probe accuracy results: maximum 2.385000, minimum 2.380000, range 0.005000, average 2.382750, median 2.382500, standard deviation 0.002077
```

Arksine, 2020-06-23:
Make sure that your x and y offsets are correct.   If they are, check for [probe location bias](https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#location-bias-check).

closed 2020-07-10 by KevinOConnor