Klipper3d/klipper #3009: Looking for best slicer for klipper

assafm25, 2020-06-20:
Hi,

Can some one help me select the best slicer + Setting for using klipper?
I use to work with Cura since i have 3d Print but for the last month i have a lots of speed problems and cant find a solution for this.

I post anther Issue with my speed problem and logs but not much results.

BlackStump, 2020-06-20:
[SuperSlicer](https://github.com/supermerill/SuperSlicer/releases)
[ideamaker](https://www.raise3d.com/download/)
[kisslicer](http://www.kisslicer.com/download-v2-alpha.html)
[craftware](https://craftbot.com/craftware/)
[icesl](https://icesl.loria.fr/download/)
A few examples there are more is you take the time to search, the best slicer is the one that works best for you, it really is a personal choice!

Hywelmartin, 2020-06-21:
Slic3r 
Prusaslicer

DeviousPenguin, 2020-06-21:
SuperSlicer has support for Klipper flavour Gcode in settings

closed 2020-06-27 by klipper-gitissuebot