Klipper3d/klipper #3010: Move exceeds maximum extrusion, print failed, sliced with Simplify3D

BTopbas, 2020-06-20:
Hello friends. My print stopped middle of print and gives this message.
```
Send: N231265 G1 X135.702 Y147.040 E0.8238*102
Recv: // Move exceeds maximum extrusion (26.738mm^2 vs 0.640mm^2)
Recv: // See the 'max_extrude_cross_section' config option for details
Recv: !! Move exceeds maximum extrusion (26.738mm^2 vs 0.640mm^2)
Changing monitoring state from "Printing" to "Cancelling"

```
I sliced object with S3D 4.1.2 and I apply Klipper special settings for S3d. My other prints no problem. I print every day. But this print stopped.

Here is picture is where print stopped. 

[image]

I searched this error on GitHub and some people's success with big values max_extrude_cross_section config. In my topic, this value must bigger than 27 I think.

But I wonder this:
Why this process not documented on slicers special settings for Klipper? 
And some people say disable "Firmware supports "sticky" parameters" in S3d g-code options. 
Which one is good for use? Example: I set max_extrude_cross_section 27 and my other print need 30 so my other print failed too. What is the best method for s3d? 

And this action needs to well document on a slicer specific settings for Klipper page. Because this will be a big print like 1-2 days and middle of the print this error not so good. 
Time and filament waste..

BTopbas, 2020-06-20:
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4808477/klippy.log.zip)

BTopbas, 2020-06-20:
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4808481/klippy.log.zip)

Hywelmartin, 2020-06-21:
30mm³/s is a rather high flow and will render a high speed with a 0.4mm nozzle.. for example.. your max speed 300mm/s  0.4 and layer height of 0.25
A normal Ender hotend can achieve 7-9 mm³/s
v6 has a max flow of 10-15mm³/s
A volcano has a max flow of 20-25mm³/s

So I think that your slicer creates weird g-code..

I use prusaslicer so I can't help you with s3d..

KevinOConnor, 2020-06-25:
An error like that is indicative of invalid g-code (a request to move an excessive amount of filament during a small move).  The error check is in place to avoid damage to the printer.  I'm not sure there's much we can say here other than it appears to be a slicer bug.

-Kevin

BTopbas, 2020-06-27:
But you know g-code with Marlin no problem. If this is a slicer bug Marlin will fail too. So klipper not compatible with s3d..

KevinOConnor, 2020-07-10:
I'm going to close this as it looks like the conversation has concluded.  (FWIW, Marlin doesn't check for those types of g-code problems, so would never raise an error.)

-Kevin

closed 2020-07-10 by KevinOConnor