Klipper3d/klipper #3015: Klipper ignoring z_offset setting [Genuine Bltouch V3]

Difusio, 2020-06-22:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4815284/klippy.log)

First of all, thanks for building a great piece of software.

Ever since I've switched out to an SKR 1.3 with Klipper it seems like Klipper is ignoring my z_offset.

My Bltouch is doing what it should do. Calibrate comes out at 2.4. Repeatability tests look fine. Pin up/down is working fine. No problems there. I've tried all possible ways of adjusting the z_offset, either through calibration or directly under bltouch section of the printer.cfg.

Every print I start with a bed_mesh level. After that it start the print, mashes the hotend into the bed and I've got to move it off the bed with the z_offset option in the tune part of the menu.

I'm kinda at a loss. Slicer is Cura with default Ender 3 profile and a G29 command added to the custom start code for bed mesh leveling.

Gilabite, 2020-06-22:
Your position_min under stepper_z section is way to low try -2. I have an ender 3 also with a similar setup and its what I have and no problems.

Difusio, 2020-06-22:
Thanks for your suggestion. That didn't help. I dit notice I had a position_endstop in my stepper_z config. I removed that, but that didn't fix it either.

Gilabite, 2020-06-22:
When you run the probe calibrate, are you saving the config afterwards? I ask because your z_offset is 0 under bltouch section, I believe that should be a positive non zero value. My z_offset from doing a probe calibrate is 1.444. That with a -2 under min_position for stepper_z. I also noticed you have 2 z motors but that they have 2 different step_distance values, shouldnt they be the same?

jakep82, 2020-06-22:
What @Gilabite said regarding step_distance.  You probably also want to use z_tilt to get your z axis aligned before running a mesh.

Difusio, 2020-06-22:
I've attached a new log with the callibrated z_offset.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4815573/klippy.log)

You're correct regarding the dual z steppers. I had picked up on that error as well and corrected it, that solved another issue that I was having; screaching noises at higher Z levels.

I'll look into z_tilt in the meantime.

With the above log I ran a print. Hotend flush against the bed again. This time I nailed the first layer by going into tune menu and giving a z_offset of +1.6. 

How would you suggest I change my callibrated value, or should that just be okay? I'm doing the paper test to callibrate it.

Gilabite, 2020-06-22:
With your current config try changing that -15 now to a -2 under stepper_z section and see if that fixes it. You have the step_distance fixed and a value under z_offset, I believe that is your problem now.

Gilabite, 2020-06-22:
I use probe_calibrate with the paper test and then save_config after. save_config will comment out whatever value you have in z_offset and place its value in the do not edit section of the config file.

jakep82, 2020-06-22:
I would do a new z_offset calibration now that your step_distance is fixed, and make sure you run save_config when you're done.

https://github.com/KevinOConnor/klipper/blob/master/docs/Probe_Calibrate.md#calibrating-probe-z-offset

Difusio, 2020-06-22:
I had allready fixed the position_min to -2. Recalibration is done and is shown in the previous log.

@jakep82 
Could you give some guidance on the z_tilt? I've got an Ender 3 with 2 Z steppers next to the x axis. Would this be a good setup? Offset irt hotend is 30x to the axis, and 90 y.

[z_tilt]
z_positions:
    -30, 90
    250, 90
points:
    60,120
    130,120
    200,120

I'll calibrate again and post a new log.

jakep82, 2020-06-22:
I'm confused about the location of your bltouch.  You're log has an x_offset of -50, and a y_offset of -16.

Difusio, 2020-06-22:
Settled on these values:

[z_tilt]
z_positions:
    -30, 90
    250, 90
points:
    60,120
    155,120
    240,120

And this is the result:

```
Send: Z_TILT_ADJUST
Recv: // probe at 60.000,120.000 is z=2.624000
Recv: // probe at 60.000,120.000 is z=2.638250
Recv: // probe at 155.000,120.000 is z=2.424500
Recv: // probe at 155.000,120.000 is z=2.431625
Recv: // probe at 240.000,120.000 is z=2.201250
Recv: // probe at 240.000,120.000 is z=2.217875
Recv: // Making the following Z adjustments:
Recv: // stepper_z = 0.343507
Recv: // stepper_z1 = -0.311097
```

Irt the bltouch position:

It's to the left of my hotend.

jakep82, 2020-06-22:
It looks like it made corrections.  Is the nozzle now at the correct Z position all the way across in the X direction?  You might also want to run z_tilt multiple times to make sure it's within an acceptable range.

Difusio, 2020-06-22:
I'll check. Here's the output for my latest calibrate:

```
Send: PROBE_CALIBRATE
Recv: // probe at 160.000,126.000 is z=2.343750
Recv: // probe at 160.000,126.000 is z=2.348500
Recv: // Starting manual Z probe. Use TESTZ to adjust position.
Recv: // Finish with ACCEPT or ABORT command.
Recv: // Z position: ?????? --> 7.346 <-- ??????
Recv: ok
[...]
Send: TESTZ Z=-6
Recv: // Z position: ?????? --> 1.346 <-- 7.346
Recv: ok
[...]
Send: TESTZ Z=-.2
Recv: // Z position: ?????? --> 1.147 <-- 1.346
Recv: ok
[...]
Send: TESTZ Z=-.2
Recv: // Z position: ?????? --> 0.947 <-- 1.147
Recv: ok
[...]
Send: TESTZ Z=-.2
Recv: // Z position: ?????? --> 0.748 <-- 0.947
Recv: ok
[...]
Send: TESTZ Z=-.2
Recv: // Z position: ?????? --> 0.548 <-- 0.748
Recv: ok
[...]
Send: TESTZ Z=-.2
Recv: // Z position: ?????? --> 0.349 <-- 0.548
Recv: ok
[...]
Send: TESTZ Z=-.2
Recv: // Z position: ?????? --> 0.149 <-- 0.349
Recv: ok
[...]
Send: TESTZ Z=-.2
Recv: // Z position: ?????? --> -0.050 <-- 0.149
Recv: ok
[...]
Send: TESTZ Z=-.2
Recv: // Z position: ?????? --> -0.250 <-- -0.050
Recv: ok
[...]
Send: TESTZ Z=.1
Recv: // Z position: -0.250 --> -0.150 <-- -0.050
Recv: ok
[...]
Send: TESTZ Z=.3
Recv: // Z position: -0.050 --> 0.149 <-- 0.349
Recv: ok
[...]
Send: ACCEPT
Recv: // bltouch: z_offset: 2.197
Recv: // The SAVE_CONFIG command will update the printer config file
Recv: // with the above and restart the printer.
Recv: ok
[...]
Send: SAVE_CONFIG
Recv: ok
Recv: // Klipper state: Disconnect
```

jakep82, 2020-06-22:
Your z_offset likely shrank because your z and z1 motors were way out of sync.

Difusio, 2020-06-22:
I"ll run a test print.

Difusio, 2020-06-22:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4815677/klippy.log)

Still no succes. Again I need to "tune" the z_offset to 1.6ish before it can extrude filament.

jakep82, 2020-06-22:
What I would do is make sure the X axis is perfectly perpendicular to the Z axis, then manually level the bed so it's parallel to the X axis.  Only after that would I attempt to calibrate the z_offset.

jakep82, 2020-06-22:
At this point you might have a parallelogram, and you'll never get it printing correctly until that's fixed.

Difusio, 2020-06-22:
I'll get on that tomorrow. Thanks for you help so far, I'll update this tomorrow.

closed 2020-07-10 by KevinOConnor